thonny/thonny #2394: shell is not visible and output is not coming

santhivanke, 2022-08-17:


aivarannamaa, 2022-08-17:
If you don't see the shell view, then you may have accidentally closed it. Try selecting "Shell" from the "View" menu.

kcarnold, 2022-09-02:
I've seen several students end up with only the editor window displayed (no Shell or anything else). I don't know how they got there, but I don't recall this being an issue in previous years.

aivarannamaa, 2022-09-03:
That's strange. At the moment I don't have any ideas how to reproduce this. Please let me know when you learn more about this!

kr-g, 2022-10-01:
@kcarnold 

`I've seen several students end up with only the editor window displayed (no Shell or anything else). I don't know how they got there, but I don't recall this being an issue in previous years.`

one remark:
when double-clicking on an editor notebook / tab the editor becomes maximized and other views and buttons are hidden. only the main menu remains.

probably this is the scenario what you have described?

to revert back deselect this option from the menu -> view -> maximize view

aivarannamaa, 2022-10-02:
> when double-clicking on an editor notebook / tab the editor becomes maximized and other views and buttons are hidden. only the main menu remains.

This can explain this, indeed. Please note that this feature is only available in the expert mode (Tools => Options => General). For un-maximizing one can also press Esc.

kcarnold, 2022-10-19:
I doubt my students would have had Expert mode enabled in the first week of class. But fortunately I haven't seen this behavior again. Close the issue for now?

aivarannamaa, 2022-10-20:
I'm happy to close it :)

closed 2022-10-20 by aivarannamaa