thonny/thonny #2400: Pi Pico errors

KG-devp, 2022-08-23:
I have two Raspberry Pi Picos both of them giving me the same error without even running the code. As the error occurs straight-up after connecting the pico.

Below is the error: -
Traceback (most recent call last):
File "<stdin>", line 7, in <module>
TypeError: function takes 0 positional arguments but 2 were given
WARNING: Could not sync device's clock: function takes 0 positional arguments but 2 were given
MicroPython v1.19.1 on 2022-06-18; Raspberry Pi Pico with RP2040

Type "help()" for more information.

/////// See attached picture as well /////
[https://drive.google.com/file/d/1VbsHyK ... sp=sharing](https://drive.google.com/file/d/1VbsHyKdyoT3inQP666VePSOqmkkbQD1O/view?usp=sharing)

I have tried to reset the Picos using the below link, but have the same error.
[https://www.okdo.com/project/raspberry- ... 9960217694](https://www.okdo.com/project/raspberry-pi-pico-reset-button/?ok_ts=1659960217694)

I have taken advice from the following forum and the advice is to raise a ticket here - https://forum.micropython.org/viewtopic.php?t=12876&p=70084 - Please can you assist with this matter.

aivarannamaa, 2022-08-23:
Thank you for the report!

Which Thonny version are you using?

Can you try the newest Thonny version: https://github.com/thonny/thonny/releases/download/v4.0.0/thonny-4.0.0.exe ?

aivarannamaa, 2023-01-28:
Please try the newest Thonny version (4.0.2 at the moment) and the newest daily build of MicroPython.

closed 2023-01-28 by aivarannamaa