thonny/thonny #2403: macOS: de-duplicate options menu entry

andypiper, 2022-08-24:
Minor UX suggestion.

On macOS there is the regular (for Mac) File->Preferences / Cmd+, menu entry.
There is also a Tools->Options menu entry (which is great for non-Mac platforms, but a duplicate of the same functionality on macOS, where the settings feature is always intended to live at File->Preferences)

If this is not something that Python + TkInter forces to happen, it might be helpful to remove the Tools->Options entry as it is unnecessary (duplicated) for Mac users.

aivarannamaa, 2022-08-25:
I assume by "File->Preferences" you meant "Thonny => Preferences".

It would be possible to make an exception for macOS and skip adding "Tools => Options", but I thought this redundance would be harmless and even beneficial -- if Mac users read a tutorial mentioning "Tools => Options", they don't have to translate the instructions for their platform.

andypiper, 2022-08-25:
Yes, I did mean that, thanks 🙂 

I'm happy to go along with your reasoning here, you are right, it saves on the documentation / tutorial side. I don't have super strong feelings apart from a slight desire to make it more aligned with the native platform!

aivarannamaa, 2022-08-25:
Alright, so let's keep it as is for now.

closed 2022-08-25 by aivarannamaa