thonny/thonny #2405: The problem with the Russian language

ErikGulqanyan, 2022-08-26:
When I copy Russian text and paste it into the IDE, it works, but when I write in Russian directly in the IDE, instead of Russian letters, a question mark (?)

ErikGulqanyan, 2022-08-27:
Latest version (4.0.0). Windows 7

> Which Thonny version are you using? On which operating system?
Latest version(4.0.0). Windows 7

aivarannamaa, 2022-08-28:
In this case you must be using Python 3.8, which has older version of the UI toolkit (Tk). I'm afraid I can't help here. 

You could try installing the official Python 3.8 (https://www.python.org/ftp/python/3.8.10/python-3.8.10.exe) and see whether you can input Russian characters properly in their IDLE envrionment (which uses the same UI toolkit). If this problem doesn't happen in IDLE, then I'll investigate why it happens in Thonny.

aivarannamaa, 2022-08-31:
@narektigranyan, which platform and Thonny version are you using?

narektigranyan, 2022-08-31:
Windows 10 with Thonny 3.3

aivarannamaa, 2022-08-31:
> Windows 10 with Thonny 3.3

Please try https://github.com/thonny/thonny/releases/download/v4.0.0/thonny-4.0.0.exe

narektigranyan, 2022-08-31:
it doesn't help

aivarannamaa, 2022-08-31:
In this case you could install https://www.python.org/ftp/python/3.10.6/python-3.10.6-amd64.exe and see if this problem also happens in IDLE. If in IDLE you can input Armenian without problems, then I need to find out why it doesn't work in Thonny. If IDLE also has problems, then we need to take this problem to Tkinter developers.

narektigranyan, 2022-08-31:
I think the problem in Tkinter, the text is copied to buffer pasted normally in editor, but when filled in whit typing there are '???'

aivarannamaa, 2022-09-01:
Some related issues at CPython tracker: 

* https://github.com/python/cpython/issues/95127
* https://github.com/python/cpython/issues/90210

aivarannamaa, 2022-09-01:
I now tried inputting Russian and Armenian on IDLE 3.8 on Windows 7 and IDLE 3.11 on Windows 10.

In both cases I was able to input Russian without problems, but instead of Armenian letters only `?`-s appeared.

I'm afraid this needs to be fixed either at Tkinter or Tk level.

closed 2023-01-28 by aivarannamaa