thonny/thonny #2406: Issues with Thonny GUI on Windows 11

rbsamoht, 2022-08-26:
## Environment

- Thonny 4.0.0
- Windows 11 (64-bit)
- Python 3.10.6 (64 bit)
- Tk 8.6.12

## Issues descriptions

1. The **About Thonny** windows reports the wrong Windows version
2. The **Files** sidebar shows an error when trying to create a file
3. The GUI shows an error when trying to close some dialog boxes with the Escape key

## Steps to reproduce issue 1

- Open Thonny
- Click on **Help** menu and select **About Thonny**
- Windows 10 (64-bit) is shown on a Windows 11 computer (see the screenshot below)

[image]

## Steps to reproduce issue 2

- Open Thonny
- Show the **Files** sidebar if it is hidden by going to the **View** menu and selecting **Files**
- Right click in a white area in the **Files** sidebar and select **New file...**
- Type a file name (eg. bug.py) and click on **OK**
- An error is shown (see the screenshot below)

[image]

By looking at the error stack, it seems that `lib\site-packages\thonny\editors.py` is calling the Python instruction below:
```
with open(filename, "rb") as fp:
```
The file should not be open in read mode since it doesn't exist

## Steps to reproduce issue 3

- Open Thonny
- Show the **Files** sidebar if it is hidden by going to the **View** menu and selecting **Files**
- Right click on a .py file in the **Files** sidebar (create one if needed)
- Select **Configure .py files...**
- Press the Esc (escape) key on your keyboard
- An error is shown (see the screenshot below)

[image]

aivarannamaa, 2022-08-27:
Thank you for the report! I wish all bug reports were as thorough and clear like this! 

I fixed nr. 1. Will deal with the others later.

aivarannamaa, 2022-08-30:
All done now. Fixes will be in version 4.0.1