thonny/thonny #2408: Unable to install Raspberry Pi Pico W / Pico WH Micropython variant

BrianGilbert, 2022-08-27:
Using Thonny 4.0.0 on Mac M1 (OS Monterey 12.5.1)
Looks like the referenced file (https://micropython.org/resources/firmware/rp2-pico-w-20220824-unstable-v1.19.1-316-g8139cbcf6.uf2) no longer exists

```
Downloading from https://micropython.org/resources/firmware/rp2-pico-w-20220824-unstable-v1.19.1-316-g8139cbcf6.uf2

Traceback (most recent call last):
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/micropython/uf2dialog.py", line 407, in _perform_work
    self._download_to_the_device(download_url, size, target_dir, target_filename)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/micropython/uf2dialog.py", line 446, in _download_to_the_device
    with urlopen(req, timeout=5) as fsrc:
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/urllib/request.py", line 216, in urlopen
    return opener.open(url, data, timeout)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/urllib/request.py", line 525, in open
    response = meth(req, response)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/urllib/request.py", line 634, in http_response
    response = self.parent.error(
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/urllib/request.py", line 563, in error
    return self._call_chain(*args)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/urllib/request.py", line 496, in _call_chain
    result = func(*args)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/urllib/request.py", line 643, in http_error_default
    raise HTTPError(req.full_url, code, msg, hdrs, fp)
urllib.error.HTTPError: HTTP Error 404: Not Found
```

aivarannamaa, 2022-08-28:
Oh, I didn't know micropython.org is deleting older nightly builds. I now updated https://raw.githubusercontent.com/thonny/thonny/master/data/micropython-variants-uf2.json, which Thonny uses for proposing the downloads. Next time you open the installation dialog, it is supposed to offer today's build.

I'll decide later which would be proper fix to this situation.

JonKinM, 2022-09-04:
I want to be able to add Custom Shortcut keys. Some shortcut keys conflict

aivarannamaa, 2022-09-10:
A better solution will appear in 4.0.1. The quest will continue under #2446

closed 2022-09-10 by aivarannamaa