thonny/thonny #2409: Sort out the XXL bundle for mac

aivarannamaa, 2022-08-28:
The XXL build for Mac is more difficult now, as I need to account for both Arm and Intel targets. The bundled Python itself is universal2 build, but most packages publish separate wheels for intel and arm and when I pip-install the packages into the bundle on my Intel Mac, only Intel wheels get downloaded.

I heard that it's possible to make pip ignore published binaries and build universal2 wheels during the install. If I manage to do this, then I'll restore the XXL bundle. Or I'll start publishing separate installers for Intel and Arm.

aivarannamaa, 2022-08-29:
I had forgotten, that the issue of different architectures also touches the regular bundle -- certain dependencies of `paramiko` (used for remote interpreters) are available only as arm or intel wheels, not as universal2.

The simplest solution would be to start publishing separate Thonny installers for Intel and Arm Macs (even though bundled Python itself is universal2). This could pose some difficulties for defining Thonny's homebrew cask, though. @andypiper, you just made the PR for Thonny's cask. Do you think you can handle separate downloads for Intel and Arm for Thonny 4.0.1? I found this, which gives me the impression that such a thing should be possible: https://github.com/Homebrew/brew/pull/10683/files 

I also considered bundling both arm and intel binaries for packages without universal2 builds and having the installer sort it out which files to install, but this solution would be much more complex to implement and it would increase the installer size.

aivarannamaa, 2022-08-29:
> I also considered bundling both arm and intel binaries for packages without universal2 builds and having the installer sort it out which files to install, but this solution would be much more complex to implement and it would increase the installer size.

I could use this approach if I ditched the XXL build for Mac, as the regular build has only couple of problematic dependencies and it's not too hard to update the postinstall script to overwrite a few specific files. But this approach wouldn't scale to XXL bundle.

andypiper, 2022-08-30:
I can look into the preferred options for ARM vs Intel brew cask installs, sure. Thanks for tagging me in 👍🏻

aivarannamaa, 2022-09-11:
I solved the issue for the regular installer. I replaced `cryptography` so-files with universal2 ones and made the installer swap out so-file for `cffi` in the postinstall script when run on Arm. This means, I can continue publishing single installer for both Intel and Arm macs.

I decided to skip publishing the XXL build for now.

closed 2022-09-11 by aivarannamaa