thonny/thonny #2410: thonny debugger not highlighting steps

grahamtacon, 2022-08-28:
I have tried to install thonny from thonny . org multiple times. I have a windows pc and I have installed the windows version.

Everything about the thonny applicatioon works EXCEPT the debugger step by step function. 

What am I doing wrong?

aivarannamaa, 2022-08-28:
What do yo precisely mean by debugger step-by-step function not working? Do you get an error message? Are the commands disabled on the toolbar or menu?

Olivier44FR, 2022-08-29:
Hello, it's the same for me, Thonny portable 4.0.0 + Win 10 and the debug buttons are greyed out. There is no way to debug step by step (CTRL+F5 does nothing either) it is as if the function was not installed.

aivarannamaa, 2022-08-29:
@Olivier44FR, Thonny doesn't have a debugger for MicroPython. Debugger can be used only with the "full" Python interpreter.

Olivier44FR, 2022-08-29:
OK, thank you

aivarannamaa, 2022-08-29:
@grahamtacon, were you also using MicroPython interpreter?

ammontence, 2022-08-30:
I'm having the same problem as @grahamtacon; I have installed the latest version 4.0.0 for windows from thonny.org. After installing and putting in some basic code, I can run the code fine, but pressing the "Debug current script" button just runs the code without stepping through, just as if I had pressed "Run current script". There are no errors, it just runs the script, and keeps the "Step over", "Step into", "Step out", and "Resume" greyed out. Here is the script I'm trying to step through:
days = ["Monday", "Tuesday", "Wednesday", "Thursday", "Friday"]
print(days)
for day in days:
    if day[0] == "F":
        print("PARTY!")
    else:
        print("Go to bed early.")

aivarannamaa, 2022-08-31:
@ammontence, do you have this code in named or unnamed editor? I can't reproduce this problem when the code is in named (i.e. saved) editor, but I did notice debugger doesn't work properly if the editor is not saved (#2415).

ammontence, 2022-08-31:
@aivarannamaa, it was unsaved. I saved the script, then was successfully able to step through the code. Thank you for your help.

aivarannamaa, 2022-09-01:
Good, I'll close this ticket and make the debugger work on unsaved files under #2415

closed 2022-09-01 by aivarannamaa