thonny/thonny #2414: Cannot connect to Raspberry Pico after configuring importer / uploading MicroPython

vi3x, 2022-08-30:
Hi, I am completely new to Thonny and Raspberry Pi and I am (unsuccessfully) trying to connect to the board from Thonny.

I tried uploading the MicroPython uf2 file both manually and via the Configure Interpreter dialog in Thonny.
The procedure seems to work, I get a "Done!" message, but when I close the dialog I get the following error:

```
Traceback (most recent call last):
  File "C:\Program Files (x86)\Thonny\lib\tkinter\__init__.py", line 1921, in __call__
    return self.func(*args)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\workbench.py", line 2016, in toolbar_handler
    handler(*args)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\running.py", line 511, in cmd_stop_restart
    self.restart_backend(True)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\running.py", line 640, in restart_backend
    self._proxy = backend_class(clean)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\mp_front.py", line 94, in __init__
    super().__init__(clean)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\mp_front.py", line 32, in __init__
    super().__init__(clean, running.get_front_interpreter_for_subprocess())
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\running.py", line 902, in __init__
    self._start_background_process(clean=clean)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\mp_front.py", line 138, in _start_background_process
    super()._start_background_process(clean=clean, extra_args=extra_args)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\running.py", line 1000, in _start_background_process
    self._proc = subprocess.Popen(
  File "C:\Program Files (x86)\Thonny\lib\subprocess.py", line 969, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Program Files (x86)\Thonny\lib\subprocess.py", line 1438, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
OSError: [WinError 87] Parametro non corretto
```

The same happens whenever I press the "Restart Backend" button.

I have tried uninstalling and re-installing Python and Thonny, rebooting the system, uninstalling and reinstalling the USB serial device that is created after the firmware upload, but nothing changed.

I tried looking into subprocess.py to get an idea about which parameter is not correct, but I'm farily new to this topic, so I couldn't figure out much.

Specs:
Raspberry Pi Pico
Windows 11 Pro 64-bit
Thonny 4.0.0
Python 3.10.6 64-bit
rp2-pico-20220618-v1.19.1.uf2

[configuration.txt](https://github.com/thonny/thonny/files/9452046/configuration.txt)

aivarannamaa, 2022-08-31:
Thank you for the report!

I've seen similar error reports before (#1000, #1082) but I haven't been able to reproduce this myself.

Can you run programs with regular Python (ie. after selecting "Local Python 3 - Thonny's Python" from the lower-right corner of Thonny's main window)?

aivarannamaa, 2022-08-31:
One more thing -- before you try "Local Python 3" it would be great if you sent me the frontend.log from the same folder you found configuration.ini from (Tools => Open Thonny data folder).

More precisely -- after you have received this error again, close Thonny and then take this file. Feel feel to review it to make sure it doesn't contain some information you don't want to share (eg. some folder or file names or something like this). Usually it doesn't contain anything sensitive.

vi3x, 2022-08-31:
Hi Aivar,

Thanks for looking into this. I am attaching the frontend.log that was created after receiving the error message and closing Thonny.

I then restarted it, switched over to Python 3 and ran a simple program, everything seems to work fine there.

I was thinking to debug the call at line 1418 of subprocess.py to see what are the (incorrect?) parameters fed to that function, but I'm not entirely sure how to go about it. Any ideas?

[LocalPythonTest](https://user-images.githubusercontent.com/21086744/187669681-042dc699-e1af-4382-a8fa-4e4df26baa9d.png)

[frontend.log](https://github.com/thonny/thonny/files/9461125/frontend.log)

aivarannamaa, 2022-08-31:
Thank you!

I suspect the reason was that the command line passed to subprocess was too long. 

Please check whether you have long filenames enabled: https://www.thewindowsclub.com/how-to-enable-or-disable-win32-long-paths-in-windows-11-10

I don't know yet, whether this is related to this error or which setting (enabled or not) would be better, but with this information I might be closer to reproducing this error on my machine.

vi3x, 2022-08-31:
Ok so... no luck with changing the paths too long setting. I tried turning the option both on and off and restart afterwards, but nothing much happened.

...BUT.

I tinkered with subprocess.py and have it raise an exception right before the process call in order to have a look at the parameter values. Not sure if it can help, but that's what it looks like:

`executable:		None
args:			"C:\Program Files (x86)\Thonny\python.exe" -u -B -m thonny.plugins.cpython_backend.cp_launcher C:\Users\vi3x\Desktop
env:			{ The entire content of PATH, some 5000 characters...}
creationflags:	512
startupinfo:	<subprocess.STARTUPINFO object at 0x000001EA8420F220>
cwd:			C:\Program Files (x86)\Thonny\lib\site-packages`

vi3x, 2022-08-31:
Update.

That "executable: None" and that argument string looked suspicious, so I hardcoded the values (see attachment).

[image]

The error is not popping up anymore, I can select Raspberry Pico as my interpreter but nothing happens (the shell box stays blank, there is no text at all and no >>>, as opposed to when you select Thonny's Python).

However, if I try and execute a script, I get this message:

`PROBLEM IN THONNY'S BACK-END: mainloop error (NotImplementedError).
See Thonny's backend.log for more info.

Process ended with exit code 17.Could not interrupt current process. Please wait, try again or select Stop/Restart!`

Here's the backend log
[backend.log](https://github.com/thonny/thonny/files/9463330/backend.log)

...and here the portion of now failing code
[image]

aivarannamaa, 2022-08-31:
If you don't get a prompt, then there is no hope of successfully running a script. I need to make sure there is more clear message in this case.

> env: { The entire content of PATH, some 5000 characters...}

I just read about the [maximum total size for the environment variables](https://devblogs.microsoft.com/oldnewthing/20031210-00/?p=41553). 5000 is not close to the maximum 32767, but if there are several large values, then ...

If you're ready to tweak the code, then select "Tools => Open Thonny program folder", go to lib\site-packages\thonny and open running.py. The code for creating the subprocess is here: 

https://github.com/thonny/thonny/blob/67ca685934c75225aa7210dd593ba1da5c2e5f66/thonny/running.py#L1000

(Wow, line 1000 and there is #1000 for this problem! What a nice coincidence)

You could refactor the value of env parameter into a variable (eg. `env`) and insert a small logging snippet before this line so we can investigate the environment sent to the new process. Perhaps something like this:

```python
total = 0

for name in sorted(env.keys()):
    value = env[name]
    total += len(name) + len(value)
    logger.info("%s : %r", name, value)

logger.info("Total env size: %d", total)
```

vi3x, 2022-08-31:
Thanks!

I get a bunch of key/values in the log, are  you interested in anything in particular?

Total env size: 4147
cwd: None
cmd_line: ['C:\\Program Files (x86)\\Thonny\\python.exe', '-u', '-B', 'C:\\Program Files (x86)\\Thonny\\lib\\site-packages\\thonny\\plugins\\rp2040\\rp2040_back.py', "{'clean': False, 'port': 'COM3', 'dtr': None, 'rts': None, 'submit_mode': 'raw_paste', 'interrupt_on_connect': True, 'write_block_size': 64, 'write_block_delay': None, 'proxy_class': 'RaspberryPiPicoBackendProxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}"]

aivarannamaa, 2022-08-31:
> are you interested in anything in particular?

no, I was interested in the total env size.

As I don't have any more ideas, you could try whether reducing the size of PATH variable changes anything. For this you need to close Thonny, go to Windows User or System environment variables dialog and reduce the PATH as much as possible (certain items may need to stay for normal operation). Of course, you should back-up the old value first. 

Does the problem still occur after this?

aivarannamaa, 2022-08-31:
... I'm suggesting this because you meantioned you have very large value in PATH variable

vi3x, 2022-08-31:
I don't think the problem is the path variable, tbh.

The "Parameter not correct" error seems to be solved by fixing this:
`executable: None
args: "C:\Program Files (x86)\Thonny\python.exe" -u -B -m thonny.plugins.cpython_backend.cp_launcher C:\Users\vi3x\Desktop`

I'm not sure if that executable is supposed to be null... but if you set it to "C:\Program Files (x86)\Thonny\python.exe", that error appears to be fixed.

The next error happens in mainloop, do you see anything weird in this backend log?

22:27:58.420 INFO    thonny: Thonny version: 4.0.0
22:27:58.420 INFO    thonny: cwd: C:\Program Files (x86)\Thonny
22:27:58.420 INFO    thonny: original argv: ['C:\\Program Files (x86)\\Thonny\\python.exe', '-u', '-B', '-m', 'thonny.plugins.cpython_backend.cp_launcher', 'C:\\Users\\vi3x\\Desktop']
22:27:58.421 INFO    thonny: sys.executable: C:\Program Files (x86)\Thonny\python.exe
22:27:58.421 INFO    thonny: sys.argv: ['C:\\Program Files (x86)\\Thonny\\lib\\site-packages\\thonny\\plugins\\cpython_backend\\cp_launcher.py', 'C:\\Users\\vi3x\\Desktop']
22:27:58.421 INFO    thonny: sys.path: ['C:\\Program Files (x86)\\Thonny', 'C:\\Program Files (x86)\\Thonny\\python310.zip', 'C:\\Program Files (x86)\\Thonny\\DLLs', 'C:\\Program Files (x86)\\Thonny\\lib', 'C:\\Program Files (x86)\\Thonny', 'C:\\Program Files (x86)\\Thonny\\lib\\site-packages']
22:27:58.421 INFO    thonny: sys.flags: sys.flags(debug=0, inspect=0, interactive=0, optimize=0, dont_write_bytecode=1, no_user_site=0, no_site=0, ignore_environment=0, verbose=0, bytes_warning=0, quiet=0, hash_randomization=1, isolated=0, dev_mode=False, utf8_mode=0, warn_default_encoding=0)
22:27:58.421 INFO    thonny: TIME/MODS 0.002 Before constructing backend (+2 modules)
22:27:58.421 INFO    thonny: NEW MODS ['encodings.ascii', 'faulthandler']
22:27:58.421 INFO    thonny: TIME/MODS 0.000 Before MainBackend 
22:27:58.421 INFO    thonny: TIME/MODS 0.000 After MainBackend 
22:27:58.422 INFO    thonny.backend: Loading plugins
22:27:58.422 INFO    thonny: TIME/MODS 0.001 Before loading plugins 
22:27:58.422 INFO    thonny.common: Using THONNY_FRONTEND_SYS_PATH ['C:\\Program Files (x86)\\Thonny\\python310.zip', 'C:\\Program Files (x86)\\Thonny\\DLLs', 'C:\\Program Files (x86)\\Thonny\\lib', 'C:\\Program Files (x86)\\Thonny', 'C:\\Program Files (x86)\\Thonny\\lib\\site-packages', 'C:\\Users\\vi3x\\AppData\\Roaming\\Thonny\\plugins\\Python310\\site-packages']
22:27:58.428 INFO    thonny.backend: Loading plugin 'thonny.plugins.backend.birdseye_backend'
22:27:58.429 INFO    thonny.backend: Loading plugin 'thonny.plugins.backend.dock_user_windows_backend'
22:27:58.430 INFO    thonny.backend: Loading plugin 'thonny.plugins.backend.flask_backend'
22:27:58.431 INFO    thonny.backend: Loading plugin 'thonny.plugins.backend.matplotlib_backend'
22:27:58.432 INFO    thonny.backend: Loading plugin 'thonny.plugins.backend.pgzero_backend'
22:27:58.433 INFO    thonny: TIME/MODS 0.011 After loading plugins (+7 modules)
22:27:58.433 INFO    thonny: NEW MODS ['pkgutil', 'thonny.plugins.backend', 'thonny.plugins.backend.birdseye_backend', 'thonny.plugins.backend.dock_user_windows_backend', 'thonny.plugins.backend.flask_backend', 'thonny.plugins.backend.matplotlib_backend', 'thonny.plugins.backend.pgzero_backend']
22:27:58.433 INFO    thonny: TIME/MODS 0.000 Beginning of mainloop 
22:28:07.262 ERROR   thonny.backend: mainloop error
Traceback (most recent call last):
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\backend.py", line 70, in mainloop
    msg = self._fetch_next_incoming_message(timeout=0.01)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\cpython_backend\cp_back.py", line 144, in _fetch_next_incoming_message
    self._read_one_incoming_message()
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\backend.py", line 151, in _read_one_incoming_message
    self._handle_immediate_command(msg)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\cpython_backend\cp_back.py", line 209, in _handle_immediate_command
    raise NotImplementedError()
NotImplementedError
22:28:07.264 INFO    thonny: TIME/MODS 8.831 Sending message ProgramOutput 
22:28:07.264 INFO    thonny: TIME/MODS 0.000 Sending message ProgramOutput 
22:28:07.264 INFO    thonny.backend: After mainloop

aivarannamaa, 2022-09-01:
I now added the explicit `executable` argument to the subprocess creation call. Please try replacing with your running.py with https://raw.githubusercontent.com/thonny/thonny/master/thonny/running.py to make sure we're on the same page.

On my computer this move didn't change anything in Thonny's behavior. Please let me know how does it work on your computer.

I'm surprised about the error you got -- the log seems to be from the Python 3 back-end, not from the MicroPython one.

aivarannamaa, 2023-01-28:
Please try, whether Thonny 4.0.2 has fixed your issue.