thonny/thonny #2419: Webrepl autocomplete does not work for variables

wohltat, 2022-09-01:
At first, i tried thonny and i like it as far as i've tried it. Very promising project.

Now to the issue.
Tab autocompletion does not work for variables although the variables are known to thonny since they appear in the 'Variables' panel on the right. 
Autocomplete seems to work for keywords though.

---------------
Version 4.0.0
OS: manjaro (arch)

aivarannamaa, 2022-09-02:
Thank you for the report!

> Tab autocompletion does not work for variables although the variables are known to thonny since they appear in the 'Variables' panel on the right.

I assume you meant autocompletion in the Shell (because autocompletion in the editor only cares about variable definitions in the written code). Anyway, I've met similar problems both in the editor and shell and I'll try to fix them for the next release.

wohltat, 2022-09-02:
Btw in the editor the tab completion does not work like in the Shell.
It just inserts a tab character. 

What about doing it like most other editors and interpreting the tab as completion key if there is text left of the cursor.

aivarannamaa, 2022-09-03:
> What about doing it like most other editors and interpreting the tab as completion key if there is text left of the cursor.

This behavior can be turned on in "Tools => Options => Editor"

wohltat, 2022-09-03:
Oh, must have missed it.
Awesome its implemented already.

I vote for making this default.

aivarannamaa, 2022-09-04:
Let's keep it open until I've fixed the problem with completing the variables.

wohltat, 2022-09-04:
Yeah right. I just confused it with the Tab-completion issue i mixed in, which is solved now.

aivarannamaa, 2023-04-02:
Thonny 4.1 will support name completions in the MicroPython / CircuitPython shell (again).

closed 2023-04-02 by aivarannamaa