thonny/thonny #2420: "Could not interrupt current process" when re-running code running a tkinter event loop

kcarnold, 2022-09-02:
1. In a new Thonny window, type `import turtle; turtle.exitonclick()`.
2. Press Run.
3. Press Run again, without first clicking on the window or pressing Stop.

Shell hangs for a moment, then says "Could not interrupt current process. Please wait, try again or select Stop/Restart!".

Stop/Restart does work. But why should pressing Run be any different than Stop then Run?

Also, while trying to reproduce this, I once got:

```
>>> %Run -c $EDITOR_CONTENT
Traceback (most recent call last):
  File "/usr/local/Caskroom/miniconda/base/envs/fa22/lib/python3.10/runpy.py", line 187, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/usr/local/Caskroom/miniconda/base/envs/fa22/lib/python3.10/runpy.py", line 110, in _get_module_details
    __import__(pkg_name)
  File "/usr/local/Caskroom/miniconda/base/envs/fa22/lib/python3.10/site-packages/thonny/__init__.py", line 1, in <module>
    import logging
  File "/usr/local/Caskroom/miniconda/base/envs/fa22/lib/python3.10/logging/__init__.py", line 26, in <module>
    import sys, os, time, io, re, traceback, warnings, weakref, collections.abc
  File "/usr/local/Caskroom/miniconda/base/envs/fa22/lib/python3.10/re.py", line 124, in <module>
    import enum
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 879, in exec_module
  File "<frozen importlib._bootstrap_external>", line 1012, in get_code
  File "<frozen importlib._bootstrap_external>", line 672, in _compile_bytecode
KeyboardInterrupt

Process ended with exit code -2.Could not interrupt current process. Please wait, try again or select Stop/Restart!
```

That was on the very first time I'd tried to run the code (did I accidentally click twice in a row really fast?), and it had a typo... but I pasted the same code again in a new window and the problem didn't recur, so I don't know what's going on with that. Actually: if I double-click the Run button really fast, or press F5 repeatedly, I can sometimes get a big traceback like that to appear.

aivarannamaa, 2022-09-03:
Thank you for the report!

Stop/Restart button kills the back-end process, but Run command tries to interrupt the current process first. I'll try to understand why is it hard to interrupt a turtle program. It may make sense to kill the program also in this situation if the interrupt doesn't work in couple of seconds.

aivarannamaa, 2022-09-11:
The Run command in Thonny 4.0.1 will kill the old process if it doesn't respond to interrupt in 1 second.

closed 2022-09-11 by aivarannamaa