thonny/thonny #2421: Unable to connect to COM3: could not open port 'COM3': PermissionError(13, 'Access is denied.', None, 5)

NoIDidntHackU, 2022-09-03:
Every time I try to run the raspberry pi Pico W with thonny it throws up this error, I have tried everything from completely resetting the Pico to going into the list of ports on my PC and resetting COM3, nothing is fixing it, no idea why.

Then when I do fix it, I get this error:
Unable to connect to COM3: port not found

Process ended with exit code 1.

aivarannamaa, 2022-09-03:
Thank you for the report!

> resetting COM3

How do you do this? It may be that another port will be assigned to the device after this.

NoIDidntHackU, 2022-09-03:
I am on Windows, which seems to be the only one that this happens to me on,
as I used it on a mac book and no worries, but for some reason it doesn't
work on Windows, the way I did it was to go into drives manager and to
reload the com3 by plugging in a usb stick I made that overloads the port,
the system then basically deletes that port to prevent a ddos attack (its
the antivirus I'm using) then all I have to do is plug something else in
and it reloads the port

NoIDidntHackU, 2022-09-03:
I basically just removed it from the device manager directory then added it back again

NoIDidntHackU, 2022-09-03:
I basically just deleted it from the device manager repository and then re added it

NoIDidntHackU, 2022-09-03:
> Thank you for the report!
> 
> > resetting COM3
> 
> How do you do this? It may be that another port will be assigned to the device after this.

I basically just deleted it from the device manager repository and then re added it

NoIDidntHackU, 2022-09-03:
> > Thank you for the report!
> > > resetting COM3
> > 
> > 
> > How do you do this? It may be that another port will be assigned to the device after this.
> 
> I basically just deleted it from the device manager repository and then re added it

I think that it's a problem with my pc as sometimes I get it on other proofreader as well, such as the coding platform for arduino

aivarannamaa, 2022-09-04:
I see. In this case I guess I can close this issue.

closed 2022-09-04 by aivarannamaa

NoIDidntHackU, 2022-09-04:
Yup