thonny/thonny #2424: typing in shell only works if cursor is at a certain position

wohltat, 2022-09-03:
If the cursor is not at the prompt than an error sound appears and nothing else happens.

I think it would be better that the cursor automatically jumps to the correct position so you can just type without worrying about moving the cursor.

aivarannamaa, 2022-09-04:
Thank you for the suggestion! I agree this would be much nicer.

wohltat, 2023-02-18:
Also if there are multiple line the command is only executed if the cursor is at the end.
I recommend a behavior like in ipython, that is 
alt + Enter : allways execute command (single or multiline) regardless of cursor position

Another key for introducing multilines would be usefull as well. Maybe not "ctrl + o" like in ipython, but ctrl + enter / shift + enter.