thonny/thonny #2425: After closing an editor in macOS, the previous editor appears without text and scrollbar

aivarannamaa, 2022-09-04:
... only the gutter remains visible.

It looks like since Tk 8.6.11, the CodeView widgets get grid_removed after creating new editor. The gutter will be re-added because of other events.

aivarannamaa, 2022-09-04:
Could not find the reason. Added forced re-gridding of the widgets as a work-around.

aivarannamaa, 2023-02-01:
@markmcgookin, please try version 4.0.2

markmcgookin, 2023-02-01:
Just updated. Exact same thing on 4.0.2

[image]

aivarannamaa, 2023-02-15:
I could now reproduce this, but I still don't know why this happens. I extended my previous hack so that it should work better now. The fix will be in 4.0.3

closed 2023-02-15 by aivarannamaa