thonny/thonny #2426: Raspberry Pi Pico does not mount

ottlukas, 2022-09-04:
Internal Error

```
Traceback (most recent call last):
  File "/app/lib/python3.10/site-packages/tkinter/__init__.py", line 1705, in __call__
    return self.func(*args)
  File "/app/lib/python3.10/site-packages/tkinter/__init__.py", line 749, in callit
    func(*args)
  File "/app/lib/python3.10/site-packages/thonny/workdlg.py", line 175, in _keep_updating_ui
    self.update_ui()
  File "/app/lib/python3.10/site-packages/thonny/plugins/micropython/uf2dialog.py", line 92, in update_ui
    targets = self.find_targets()
  File "/app/lib/python3.10/site-packages/thonny/plugins/micropython/uf2dialog.py", line 121, in find_targets
    for vol in list_volumes(skip_letters=["A"])
  File "/app/lib/python3.10/site-packages/thonny/misc_utils.py", line 98, in list_volumes
    mount_output = subprocess.check_output("mount").splitlines()
  File "/usr/lib/python3.10/subprocess.py", line 420, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.10/subprocess.py", line 501, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/usr/lib/python3.10/subprocess.py", line 969, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.10/subprocess.py", line 1845, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'mount'
```
Thonny Version: 4.0.0 (installed via Flathub)
Operating System: Linux Mint 21                   
Kernel: Linux 5.15.0-47-generic
Python 3.10.6 (64 bit)
Tk 8.6.12

```
Couldn't find the device automatically. 
Check the connection (making sure the device is not in bootloader mode) or choose
"Configure interpreter" in the interpreter menu (bottom-right corner of the window)
to select specific port or another interpreter.
```
Anyone has an idea how I can debug / fix this internal errror in Thonny to flash my Raspberry Pi Pico? - Thank you!

ottlukas, 2022-09-04:
Ok the issue was cable related see also: https://support.thepihut.com/hc/en-us/articles/360017234598-My-computer-won-t-detect-my-Raspberry-Pi-Pico

aivarannamaa, 2022-09-04:
The error trace said Thonny was not able to execute Linux command `mount`, which it uses for finding your Pico's bootloader drive.

Did replacing the cable fix the problem for you?

ottlukas, 2022-09-04:
I switched to Windows and there Thonny worked for me.
Thonny 4.0.0
Windows 10
Python 3.10.6
Tk 8.6.12

I ll keep you posted if i try it again on Linux.

aivarannamaa, 2022-09-04:
When you are back in Linux, then please check what does `which mount` give you in Terminal.

ottlukas, 2022-09-04:
Terminal output:
```
which mount
/usr/bin/mount
```

fdisk -l gives me:
```
Festplatte /dev/sdb: 128 MiB, 134217728 Bytes, 262144 Sektoren
Festplattenmodell: RP2             
Einheiten: Sektoren von 1 * 512 = 512 Bytes
Sektorgröße (logisch/physikalisch): 512 Bytes / 512 Bytes
E/A-Größe (minimal/optimal): 512 Bytes / 512 Bytes
Festplattenbezeichnungstyp: dos
Festplattenbezeichner: 0x00099910
```

So Linuxmint can mount the RP but it ends in the shown error...

jwillikers, 2023-02-26:
This appears to be the first appearance of the issues flathub/org.thonny.Thonny#153 and #2650. Using UDisks2 to mount filesystems via its DBUS API should resolve this.

aivarannamaa, 2023-07-05:
Should be fixed by now.

closed 2023-07-05 by aivarannamaa