thonny/thonny #2428: Plugins seem not to be in path and so not available to each other

ettore-galli, 2022-09-04:
When  plugin requires another one it may fail because python executable and plugin directory are in different path.

This happens for example in my machine and with the plugins:

thonny-black-format
black

Both have been installed via plugin manager but when thonny-black-format runs it does not find the module "black"  on its path.

Reading around this seems to be due to the fact that Thonny 4.0.0 separates plugins path from the others.

Is there some workaround?

aivarannamaa, 2022-09-04:
Thonny 4 has a separate directory, where the plug-in manager (Tools => Manage plug-ins) installs plug-in modules.

Before Thonny's GUI process starts loading the plug-ins, it adds this directory to `sys.path`.

For the back-end plug-ins, it's almost the same, but in this case the extra directory is removed from the `sys.path` after loading the plug-ins.

`thonny-black-format` plug-ins seems to be abandoned. At least the GitHub project linked from https://github.com/Franccisco/thonny-black-code-format is not available anymore.

ettore-galli, 2022-09-04:
Could you point me to a well done plugin so that I can adapt it to black?

aivarannamaa, 2022-09-04:
The plug-ins have very different purposes and approaches. I think it's better if you show me the code of the problematic plug-in and I'll see if I spot something to change there.

ettore-galli, 2022-09-06:
I managed to re-write the plug-in starting from the installed sources.

Now it is available at:

[https://pypi.org/project/thonny-black-formatter/](https://pypi.org/project/thonny-black-formatter/)

I suggest that the issue can be closed.

If Thonny devs find it worth,  maybe it can be added to the Thonny home page...

aivarannamaa, 2022-09-09:
Nice!

I now added thonny-black-formatter to the plugins list at Thonny home page

closed 2022-09-09 by aivarannamaa