thonny/thonny #2429: I help with a project that uses Thonny. Various Thonny bugs have come up.

RAYTRAC3R, 2022-09-04:
Hello! I'm on the development team for a fangame (based on a popular book series) made in Pygame called ClanGen. We compile the game into an executable for full releases, but whenever someone wants to try the source-code-based dev versions, we recommend using Thonny to run it! 
Many of us devs use Thonny ourselves to work on the game, myself included. I haven't personally ran into any issues, and many fans run the game in Thonny just fine, however a handful of our fans have been dealing with various errors while trying to run the game, that _seem_ to be errors with Thonny rather than our own code. Most people can run the game well, but there's been enough of these errors that I figured I'd try to contact you guys to see if any issues can be identified from this!
### Case One
A user tries to launch the game, and gets this error: 
`Traceback (most recent call last):
  File "C:\Users\jrrac\AppData\Local\Programs\Thonny\lib\site-packages\thonny\shell.py", line 1273, in _submit_input
    get_shell().set_title("")
  File "C:\Users\jrrac\AppData\Local\Programs\Thonny\lib\site-packages\thonny\shell.py", line 159, in set_title
    notebook.tab(container, text=title)
  File "C:\Users\jrrac\AppData\Local\Programs\Thonny\lib\tkinter\ttk.py", line 896, in tab
    return _val_or_dict(self.tk, kw, self._w, "tab", tab_id)
  File "C:\Users\jrrac\AppData\Local\Programs\Thonny\lib\tkinter\ttk.py", line 298, in _val_or_dict
    res = tk.call(*(args + options))
_tkinter.TclError: .!frame.!automaticpanedwindow.!automaticpanedwindow2.!automaticnotebook.!frame is not managed by .!frame.!automaticpanedwindow.!automaticpanedwindow2.!automaticnotebook`
[image]
[I did some research for them, and found a similar error mentioned on the Thonny Github, but no fixes were listed.](https://github.com/thonny/thonny/issues/2142)
They had Pygame, a required prerequisite for ClanGen, installed already, along with several optional prerequisites for the game.
They tried reinstalling twice. The first time, it didn't fix anything. The second time, they seem to have made sure to remove _all_ files related to Thonny across their entire system, and it fixed it.
### Case Two
A user tries to run the game in Thonny after updating it. Even if they try redownloading the game, they get this error, a very similar one to the previous one:
`Traceback (most recent call last):
  File "D:\Thonny\lib\site-packages\thonny\shell.py", line 1273, in _submit_input
    get_shell().set_title("")
  File "D:\Thonny\lib\site-packages\thonny\shell.py", line 159, in set_title
    notebook.tab(container, text=title)
  File "D:\Thonny\lib\tkinter\ttk.py", line 896, in tab
    return _val_or_dict(self.tk, kw, self._w, "tab", tab_id)
  File "D:\Thonny\lib\tkinter\ttk.py", line 298, in _val_or_dict
    res = tk.call(*(args + options))
_tkinter.TclError: .!frame.!automaticpanedwindow.!automaticpanedwindow2.!automaticnotebook.!frame is not managed by .!frame.!automaticpanedwindow.!automaticpanedwindow2.!automaticnotebook`
 [image]
 When they try to hit the stop button, they get a similar but slightly different error.
 `Traceback (most recent call last):
  File "D:\Thonny\lib\site-packages\thonny\workbench.py", line 1776, in event_generate
    handler(event)
  File "D:\Thonny\lib\site-packages\thonny\shell.py", line 1418, in _on_backend_terminated
    get_shell().set_title("")
  File "D:\Thonny\lib\site-packages\thonny\shell.py", line 159, in set_title
    notebook.tab(container, text=title)
  File "D:\Thonny\lib\tkinter\ttk.py", line 896, in tab
    return _val_or_dict(self.tk, kw, self._w, "tab", tab_id)
  File "D:\Thonny\lib\tkinter\ttk.py", line 298, in _val_or_dict
    res = tk.call(*(args + options))
_tkinter.TclError: .!frame.!automaticpanedwindow.!automaticpanedwindow2.!automaticnotebook.!frame is not managed by .!frame.!automaticpanedwindow.!automaticpanedwindow2.!automaticnotebook`
 [image]
 The assistant panel was always blank, and the shell said nothing (and had to be opened manually as a cmd).
Some people who got the same error mentioned reinstalling fixed it, so they tried uninstalling it (from D drive) and reinstalling it (in C drive), but got the same error. (This is possibly because they might have done an incomplete uninstall) 
They also tried various file paths and restarting their PC. They had pygame installed already, and they're on Windows 10. As far as I know, they never got the game working in Thonny again.
### Case Three
 A user updates to the latest version and tries to run it in Thonny. They get this error.
 `Traceback (most recent call last):
  File "C:\Users\säde\AppData\Local\Programs\Thonny\lib\site-packages\thonny\shell.py", line 1276, in _submit_input
    get_runner().send_command(
  File "C:\Users\säde\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 262, in send_command
    response = self._proxy.send_command(cmd)
  File "C:\Users\säde\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 1043, in send_command
    self._send_msg(cmd)
  File "C:\Users\säde\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 1050, in _send_msg
    self._proc.stdin.write(serialize_message(msg) + "\n")
OSError: [Errno 22] Invalid argument

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\säde\AppData\Local\Programs\Thonny\lib\tkinter__init.py", line 1921, in call
    return self.func(*args)
  File "C:\Users\säde\AppData\Local\Programs\Thonny\lib\tkinter__init.py", line 839, in callit
    func(*args)
  File "C:\Users\säde\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\highlight_names.py", line 55, in consider_request
    self._request()
  File "C:\Users\säde\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\highlight_names.py", line 73, in _request
    runner.send_command(
  File "C:\Users\säde\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 262, in send_command
    response = self._proxy.send_command(cmd)
  File "C:\Users\säde\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 1043, in send_command
    self._send_msg(cmd)
  File "C:\Users\säde\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 1050, in _send_msg
    self._proc.stdin.write(serialize_message(msg) + "\n")
OSError: [Errno 22] Invalid argument`
They mentioned to us that on a very recent previous version of ClanGen, the game ran in Thonny just fine (albeit with some bugs on the ClanGen end). This is the first time this specific individual had an error, but the error they had has been experienced by multiple other users.
### End
These are only three of several cases of Thonny based errors, however most other cases are very similar to these, and this is getting long already, so I figured I'd just show off these ones. 
Any insight would be appreciated, especially on Case Three because it's the most recent one and the user seems to still be having issues as we speak. 
Hopefully this helps both the ClanGen team and the Thonny team!

aivarannamaa, 2022-09-06:
Thanks a lot! I'll try to reproduce them. If I can reproduce, then I'll fix them in 4.0.1

aivarannamaa, 2022-09-09:
I have a hypothesis that in these cases Thonny's Shell pane was closed when the script was run. Thonny 4.0.0 came with a regression, that caused exception in this case (#2412).

It would be very helpful if you could confirm or refute this.

aivarannamaa, 2022-09-11:
The assumed error has been fixed in 4.0.1 (just published). Please re-open, if the error persists with 4.0.1

closed 2022-09-11 by aivarannamaa