thonny/thonny #2430: IndexError: tuple index out of range when trying to close other tabs

michelcrypt4d4mus, 2022-09-04:
error text below.
sadly i do not have thonny available right now to give recreation for 100% certain but i _believe_ the error can be reproduced like this:

1. open a bunch of tabs
2. move/delete one of the files loaded in one of those tabs
3. agree with Thonny's suggestion to close that tab
4. try to close other tabs on any open tab

```
Traceback (most recent call last):
  File "/usr/lib/python3.9/tkinter/__init__.py", line 1892, in __call__
    return self.func(*args)
  File "/home/quicklord/.local/lib/python3.9/site-packages/thonny/ui_utils.py", line 538, in _close_other_tabs
    self.close_tabs(self._popup_index)
  File "/home/quicklord/.local/lib/python3.9/site-packages/thonny/ui_utils.py", line 545, in close_tabs
    self.close_tab(tab_index)
  File "/home/quicklord/.local/lib/python3.9/site-packages/thonny/editors.py", line 891, in close_tab
    editor = self.get_child_by_index(index)
  File "/home/quicklord/.local/lib/python3.9/site-packages/thonny/ui_utils.py", line 556, in get_child_by_index
    tab_id = self.tabs()[index]
IndexError: tuple index out of range
```

aivarannamaa, 2022-09-06:
Thank you for the report!

Unfortunately I was not able to reproduce it yet, but I'll do a code review later.

aivarannamaa, 2023-07-05:
I'm closing this as Thonny 5 will get a new EditorNotebook and this error is not relevant anymore.

closed 2023-07-05 by aivarannamaa