thonny/thonny #2431: Stepping through python code button don't work in Thonny

saifalafeefi, 2022-09-05:
for some reason my stepping buttons for debugging seems to be disabled or something (I can't click on them), 

any solutions?

[image]

aivarannamaa, 2022-09-05:
Are you using a MicroPython or CircuitPython interpreter? Debugger is implemented only for the "big" Python.

saifalafeefi, 2022-09-05:
I was trying to follow a basic python tutorial (like for loop, printing, etc..), the video showed that it was possible to step into the code one at a time. But it doesn't seem to be a big of an issue rn, just curious on why's that.

aivarannamaa, 2022-09-06:
I now looked at your screenshot more carefully and noticed the Debug button (with the green bug) is enabled. The stepping commands only became available after you start the debug process by pressing on the bug button.

saifalafeefi, 2022-09-06:
> I now looked at your screenshot more carefully and noticed the Debug button (with the green bug) is enabled. The stepping commands only became available after you start the debug process by pressing on the bug button.

It still won't work, I might have a 'bugged' version of Thonny

aivarannamaa, 2022-09-06:
> It still won't work

What happens, when you press the "bug" button?

robinjuliatrute, 2022-09-06:
I don't know if its similar but I just tried to step through a snippet of python via the debug option and thonny just ignored my breakpoints, ran it as if I clicked on run. Saving the snippet fixed the issue.

solsword, 2022-09-08:
Can offer some confirmation here: I teach w/ thonny and trying the debugger when the file has not yet been saved (which is only newly possible in 4.0) doesn't seem to work. Saving the file also made it work for us.

(BTW, despite all the v4 bugs you're dealing with, thanks for making such a great beginner IDE!)

aivarannamaa, 2022-09-08:
@solsword, thank you for the encouraging words! I'll try to release 4.0.1 with fixes by the end of this week!

greyscones, 2023-07-24:
i simply double-clicked my code and it started working.