thonny/thonny #2435: Trouble with help function

sebhoa, 2022-09-06:
- OS : macOS Monterey 12.4 but someone had same issue with Ubuntu
- Installation : by pip3 command to use my already installed Python interpreter 3.9

In the Thonny console:

```
help(print)
```

no output but python seems waiting for something as with the input function, I press return and:

```
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/tkinter/__init__.py", line 1892, in __call__
    return self.func(*args)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/tkinter/__init__.py", line 814, in callit
    func(*args)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/thonny/running.py", line 548, in _poll_backend_messages
    if self._pull_backend_messages() is False:
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/thonny/running.py", line 561, in _pull_backend_messages
    msg = self._proxy.fetch_next_message()
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/thonny/plugins/cpython_frontend/cp_front.py", line 138, in fetch_next_message
    msg = super().fetch_next_message()
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/thonny/running.py", line 1218, in fetch_next_message
    if msg.event_type == "ProgramOutput":
AttributeError: 'InputSubmission' object has no attribute 'event_type'
```

Any idea?

a-anjos, 2022-09-06:
Same here.
Thonny 4.0.0
Linux 64 bit (Debian bookworm)
Python 3.10.6
TK 8.6.12

aivarannamaa, 2022-09-09:
The fix will be in 4.0.1

closed 2022-09-09 by aivarannamaa