thonny/thonny #2436: ModuleNotFoundError when using Thonny installer

JLLangy, 2022-09-06:
I have installed Thonny 4.0.0 using the MacOS installer (I'm using 12.5.1). The interpreter returns a module not found error when attempting to execute code which uses PyGame Zero. (import pgzrun ModuleNotFoundError: No module named 'pgzrun')

Subsequently, I installed IDLE. Then I installed pygame zero and pygame using PIP3. IDLE ran the code without issue.

I then used PIP to install Thonny. The PIP-installed Thonny also successfully executed the code, but the installer supplied version of Thonny still will not execute it.

Also, when attempting to use the Plug-in Manager from the PIP-installed version, almost everything I try to do hangs Thonny and requires a Force-quit.

aivarannamaa, 2022-09-07:
Do you pip-install pgzero from Thonny's package manager (Tools => Manage packages) or Thonny's environment (Tools => Open system shell) or outside of Thonny?

Please note that Thonny comes with it's own Python interpreter. If you have another Python interpreter on your system and you install packages outside of Thonny, then most likely these packages get installed for this other interpreter and won't be visible for Thonny's Python.

JLLangy, 2022-09-07:
I have tried installing using Thonny's Package Manager. This gives the ModuleNotFoundError previously stated.

Following your suggestion, I have tried Thonny's environment (Tools => Open System Shell), then used pip install pgzero.
This produces the following error when interpreting my code:

Traceback (most recent call last):
  File "/Users/[me]/Library/CloudStorage/OneDrive-Personal/[path and filename]", line 1, in <module>
    import pgzrun
  File "/Users/[me]/Library/Python/3.10/lib/python/site-packages/pgzrun.py", line 13, in <module>
    from pgzero.runner import prepare_mod, run_mod
  File "/Users/[me]/Library/Python/3.10/lib/python/site-packages/pgzero/runner.py", line 1, in <module>
    import pygame
  File "/Users/[me]/Library/Python/3.10/lib/python/site-packages/pygame/__init__.py", line 83, in <module>
    from pygame.base import *  # pylint: disable=wildcard-import; lgtm[py/polluting-import]
ImportError: dlopen(/Users/jll/Library/Python/3.10/lib/python/site-packages/pygame/base.cpython-310-darwin.so, 0x0002): tried: '/Users/[me]/Library/Python/3.10/lib/python/site-packages/pygame/base.cpython-310-darwin.so' (mach-o file, but is an incompatible architecture (have (arm64), need (x86_64)))

JLLangy, 2022-09-18:
In the lower-right it says "Local Python 3 - Thonny's Python".

I've still been unable to get pgzero to work using the "run" button, and I have tried all of the different interpreter options in Tools > Thonny Options > Interpreter tab > Python executable.

However, using Run > Run current script in Terminal will successfully execute the script.

aivarannamaa, 2022-09-19:
Do you have Intel or Arm Mac?

What does it say when you execute following in Thonny's Shell:

```
>>> import sys
>>> sys.executable
```

JLLangy, 2022-09-19:
I'm using an ARM Mac (M1 Pro).

>>> import sys
>>> sys.executable
'/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/bin/python3.10'

alfiecg24, 2022-11-20:
As @aivarannamaa said, there is a compatibility error between ARM64 and x86_64. If you haven't managed to fix this yet, open your Applications folder, right click on Thonny and select Get Info. After this, deselect 'Open using Rosetta'. For some reason, this is enabled by default - which means that macOS runs the x86_64 version of Thonny, but tries to load ARM64 libraries.

On a side note, you have to also save the Python file in Thonny before executing `import pgzrun`, or it will throw an error because it cannot be run from an active interpreter.

Hope this helped

JLLangy, 2022-11-21:
> open your Applications folder, right click on Thonny and select Get Info. After this, deselect 'Open using Rosetta'.

This has solved the issue - thank you

closed 2022-11-21 by JLLangy