thonny/thonny #2439: Thonny Fails to run Program in Flatpak Package

CapeCrusader321, 2022-09-08:
After the latest update my thonny downloaded from flathub isnt functional it gives the following dialog with error and copy to clipboard option-

Traceback (most recent call last):
  File "/app/lib/python3.10/site-packages/thonny/shell.py", line 1273, in _submit_input
    get_shell().set_title("")
  File "/app/lib/python3.10/site-packages/thonny/shell.py", line 159, in set_title
    notebook.tab(container, text=title)
  File "/app/lib/python3.10/site-packages/tkinter/ttk.py", line 903, in tab
    return _val_or_dict(self.tk, kw, self._w, "tab", tab_id)
  File "/app/lib/python3.10/site-packages/tkinter/ttk.py", line 299, in _val_or_dict
    res = tk.call(*(args + options))
_tkinter.TclError: .!frame.!automaticpanedwindow.!automaticpanedwindow2.!automaticnotebook.!frame is not managed by .!frame.!automaticpanedwindow.!automaticpanedwindow2.!automaticnotebook

CapeCrusader321, 2022-09-08:
[image]

aivarannamaa, 2022-09-08:
Thank you for the report!

I see that you have closed the Shell pane. Could you please try whether this problem still occurs after you open the the Shell (View => Shell)?

CapeCrusader321, 2022-09-08:
Strangely its working with shell opened but by default thonny doesnt open or give the shell view, doing it manually always feels bit of a tedious work, also in few programs it gives the following error-

PROBLEM IN THONNY'S BACK-END: mainloop error (FileNotFoundError: [Errno 2] No such file or directory).
See Thonny's backend.log for more info.

CapeCrusader321, 2022-09-08:
[image]

CapeCrusader321, 2022-09-08:
This problem is limited to few programs only and doesnt affect all programs. In other programs it works as follows-

[image]

aivarannamaa, 2022-09-08:
Thank you for this information!

At the moment I don't have any ideas which file Thonny doesn't find. If you can reproduce this error, then after receiving it, please select "Tools => Open Thonny data folder", close Thonny and send me the contents of backend.log (or attach the file itself).

CapeCrusader321, 2022-09-09:
Sorry for the delay, here is the backend.log file (Attached)

[backend.log](https://github.com/thonny/thonny/files/9535163/backend.log)

aivarannamaa, 2022-09-09:
Thanks! It turns out Thonny's backend is not able to compute current working directory...

I now noticed that your script is under `/run/user`. Did you save your script there on purpose or was it the first directory Thonny proposed to save files into?

CapeCrusader321, 2022-09-09:
Everything is set to default as Thonny does by default,i store my files in the home directory so i think it was the first directory Thonny proposed to save files into. I will try to provide more info if needed...

aivarannamaa, 2022-09-09:
I guess I have to learn the quirks of Flatpak Thonny myself first.

CapeCrusader321, 2022-09-11:
Maybe!,waiting for the fix, as a Satisfactory measure i purged the Flatpak Thonny with its all configurations and data and reinstalled it from Flathub and even without any major tweaks in thonny settings,the Problem continues to persist.

aivarannamaa, 2022-09-11:
The first issue was fixed in 4.0.1 (just published). I'll investigate the second one later. Please let me know if you find a way for reproducing the issue.

CapeCrusader321, 2022-09-13:
Hey after a few tests on my level its confirmed that  yes the first issue doesnt exist any more but the second one is still there this time with a slightly different output,Check it out-

[image]

CapeCrusader321, 2022-09-13:
Another interesting thing to note- The Backend log after this error doesnt mention any error unlike previous time or version, Here is the backend log-

[backend.log](https://github.com/thonny/thonny/files/9559183/backend.log)

CapeCrusader321, 2022-09-13:
And remains the Reproduction of this issue i request you to test my Program in a flatpak,Thonny package on a Ubuntu 20.04.5 Machine having latest ppa installed from flatpak ppa. Or you can try these program in your own Distro with flatpak and see if the problem is specific to Ubuntu machines or platform independent.
Here is my Program that Is giving this error-

[ErrorCode2.txt](https://github.com/thonny/thonny/files/9559414/ErrorCode2.txt)

CapeCrusader321, 2022-09-13:
New Discovery- A new file created with the Same program runs successfully but the old file still gives error, i will experiment more with changing the problematic file's location and producing error.

jwillikers, 2022-12-17:
@aivarannamaa @CapeCrusader321 The Flatpak shows files under `/run/user` when they are mounted in from the host filesystem inside the Flatpak via the sandbox. I don't think it's possible to mount in an entire directory via the Thonny Flatpak at the moment, but I might be able to figure out a better way to handle this.

jwillikers, 2023-02-26:
@CapeCrusader321 Do you still get this error using the latest Thonny Flatpak? I think it should be fixed now.

CapeCrusader321, 2023-02-27:
Hey its been a long time, i will checkout, its been months since i used the Flatpak version, now i am working on creating and maintaining snap package for Thonny.

closed 2024-02-21 by capecrusader-121