thonny/thonny #2444: Thonny runs Raspberry code on Shell, but not on editor

josegmp, 2022-09-09:
I am using Thonny 4.0.0 to write Micropython (1.19.1) code for a Raspberry Pi Pico W. I am new to Thonny and Raspberry, and as a first experiment, am trying to scan for all available networks, so I can choose the one with the better signal and connect to it.
Problem is that the code will not run on the content editor, but when I copy/paste it to the Shell line by line, it runs without issue.

Here I try to run the code, and I get an error on line 33. The problem starts on line 13, when the "scan" never occurs.
[image]

"available_wlan" is empty
[image]

If I copy/paste the first several lines into the shell, it works well
[image]

Any advice appreciated.

aivarannamaa, 2022-09-09:
The most important part here is that you have syntax error on line 33 (else-branch without anything in it). If you run the code all together, then MicroPython first tries to parse it, but because of the syntax error it gives up.

If you feed the code piece by piece, then MicroPython is happy to execute all pieces which have correct syntax.

josegmp, 2022-09-09:
Hello Aviar. I'll complete it and then give it another try. Issue closed.
Best regards.

closed 2022-09-09 by josegmp