thonny/thonny #2451: I'm having the same problem as @grahamtacon; I have installed the latest version 4.0.0 for windows from thonny.org. After installing and putting in some basic code, I can run the code fine, but pressing the "Debug current script" button just runs the code without stepping through, just as if I had pressed "Run current script". There are no errors, it just runs the script, and keeps the "Step over", "Step into", "Step out", and "Resume" greyed out. Here is the script I'm trying to step through:

leadbewj, 2022-09-11:
Having the same problem, Installed from 4.0, I have saved the file and reopened and still do not get the "step Over"  etc.  What am i doing wrong
I'm having the same problem as @grahamtacon; I have installed the latest version 4.0.0 for windows from thonny.org. After installing and putting in some basic code, I can run the code fine, but pressing the "Debug current script" button just runs the code without stepping through, just as if I had pressed "Run current script". There are no errors, it just runs the script, and keeps the "Step over", "Step into", "Step out", and "Resume" greyed out. Here is the script I'm trying to step through:

```python
days = ["Monday", "Tuesday", "Wednesday", "Thursday", "Friday"]
print(days)
for day in days:
    if day[0] == "F":
        print("PARTY!")
    else:
        print("Go to bed early.")
```

_Originally posted by @ammontence in https://github.com/thonny/thonny/issues/2410#issuecomment-1231778631_

leadbewj, 2022-09-11:
Never mind.  I found the issue.  Under Tools >> Options >> "Preferred debugger"  needs to be set to faster.

aivarannamaa, 2022-09-11:
Actually, a simpler work-around is to save the file first (I'm assuming your script wasn't saved to a file when you started debugger). See #2415.

Please re-open if you can't debug a saved script.

closed 2022-09-11 by aivarannamaa