thonny/thonny #2454: AttributeError when saving (sometimes)

JPZV, 2022-09-13:
Sometimes when I try to save a file (either locally or remotely to my RPi Pico) I get this exception:

```
Traceback (most recent call last):
  File "D:\Program Files (x86)\Thonny\lib\site-packages\thonny\workbench.py", line 1787, in event_generate
    handler(event)
  File "D:\Program Files (x86)\Thonny\lib\site-packages\thonny\running.py", line 1514, in _on_response
    if response.get("command_id") == getattr(self._cmd, "id"):
AttributeError: 'InlineCommand' object has no attribute 'id'
```

Step to reproduce:

1) Connect RPi Pico and open Thonny
2) Open a .py file
3) Make a change
4) Press Ctrl + S
5) Repeat steps 3 and 4 until the exception appears

[image]

**OS**: Windows 11 21H2 (Build 22000.856)
**Device**: Raspberry Pi Pico (non-W)
**Firmware**: MicroPython v1.18-10-g5db278f1d-dirty on 2022-05-10
**Connected by**: MicroUSB through my Keyboard (dedicated USB Port connected directly to my Motherboard)

aivarannamaa, 2022-09-15:
Thank you for the report!

I'll try to reproduce and fix it later

thatvoltguy, 2022-10-06:
Can confirm; seeing on macos as well:
`Traceback (most recent call last):
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/workbench.py", line 1787, in event_generate
    handler(event)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/running.py", line 1514, in _on_response
    if response.get("command_id") == getattr(self._cmd, "id"):
AttributeError: 'InlineCommand' object has no attribute 'id'`

aivarannamaa, 2022-12-29:
The fix will be in the upcoming 4.0.2.

closed 2022-12-29 by aivarannamaa