thonny/thonny #2458: Debugger has some broken graphics

isaacl, 2022-09-14:
The last step before the debugger invokes a function stays on the screen, over the top of the new function pop up.

[image]

It also shows even when thonny is not the top window :-o (system: OS X)

[image]

Sample code:
```python
def f2(v):
    return v

def main():
    f2(2)

main()
```

aivarannamaa, 2022-09-19:
Thank you for this report!

I'll try to reproduce and fix this later.

isaacl, 2022-09-19:
My students with Windows did not see this issue, FWIW.   I think it is OS X specific.

aivarannamaa, 2023-07-05:
I can't reproduce this in Thonny 4.1.1. I assumed it is fixed by now. Please reopen if you still have this problem with latest Thonny.

closed 2023-07-05 by aivarannamaa