thonny/thonny #2460: Device is busy or does not respond.

JawadMezaal, 2022-09-16:
I have installed Thonny to work on ESP32-WROOM-32. I downloaded the firmware from Micropython, but all the time I got this message. I tried to stop and run but all the time no response.
I connected to different computers and change the cable, but it still isn't working.
Please, do you have any Idea to solve this problem?

aivarannamaa, 2022-09-19:
Thank you for the report!

Did you try Ctrl-C?

Can you get to REPL with some simpler tools (eg. Putty or tio)?

JawadMezaal, 2022-09-19:
Thank you for replying,
I tried all the recommendations in the error message but got no response. I
did not try with putty or to. I think need to learn them and try them.

closed 2023-01-28 by aivarannamaa