thonny/thonny #2462: This started happening today. I Tried reinstalling Python3.10 and Thonny. Thonny opens, and immediately i get this.

elemenel, 2022-09-17:
AttributeError: module 'enum' has no attribute 'IntFlag'
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 62, in apport_excepthook
    if not enabled():
  File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 24, in enabled
    import re
  File "/usr/lib/python3.10/re.py", line 145, in <module>
    class RegexFlag(enum.IntFlag):
AttributeError: module 'enum' has no attribute 'IntFlag'

Original exception was:
Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 187, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/usr/lib/python3.10/runpy.py", line 110, in _get_module_details
    __import__(pkg_name)
  File "/home/elemen/.local/lib/python3.10/site-packages/thonny/__init__.py", line 1, in <module>
    import logging
  File "/usr/lib/python3.10/logging/__init__.py", line 26, in <module>
    import sys, os, time, io, re, traceback, warnings, weakref, collections.abc
  File "/usr/lib/python3.10/re.py", line 145, in <module>
    class RegexFlag(enum.IntFlag):
AttributeError: module 'enum' has no attribute 'IntFlag'

Process ended with exit code 1.

elemenel, 2022-09-17:
I was able to resolve this issue by deleting the files referencing Thonny in the '/usr/lib folder. I then installed Thonny into a virtual environment. Everything is working again. And thanks for the recent updates to Thonny. I like them! 

I am a retiree, taking up Thonny as a hobby.

closed 2022-09-17 by elemenel

aivarannamaa, 2022-09-19:
Thank you for the report!

There is a standard library module named `enum` and your stacktrace says another module doesn't find expected attributes in this?

Did you, by chance, create a script named `enum.py` in your working directory?

elemenel, 2022-09-19:
Hi, no script named enum.py was created by me. The attribute error and exception occurred immediately, during initialization after I opened Thonny. So I am thinking that the issue was in Python, and not Thonny.