thonny/thonny #2465: debub problem

elstarkov, 2022-09-18:
'PROBLEM IN THONNY'S BACK-END: Exception while handling 'Debug' (AttributeError: module 'numbers' has no attribute 'Number').
See Thonny's backend.log for more info.

Process ended with exit code 1.'

Any idea what I should do to fix this? It appears when I try do debug.

aivarannamaa, 2022-09-19:
Thank you for the report!

Which Thonny version do you use?

Do yo have a script named `numbers.py` in your working directory? It may shadow a built-in module with the same name.

elstarkov, 2022-09-19:
I used the most recent one, 4,2 if I remember correctly. Then I got the tip to try a different version such as 3.1, the one from november 2019, and now it's working again. I think I'll just stick to that version. Thank you anyway!

closed 2023-01-28 by aivarannamaa