thonny/thonny #2466: [Feature] Suggest to activate `venv` and `.venv` -containing folders

Ivoz, 2022-09-19:
I was very happy with this feature, 

> Add possibility to activate a virtual environment via Files view (by @kr-g), https://github.com/thonny/thonny/pull/2381

However most of my virtual env's live in a `.venv` folder, which unfortunately is hidden by Thonny. So I can never activate those with this feature. It would be cool if a UI label was added to any folders that contain a `venv` / `.venv` folder inside them, and to have the ability to "activate" such a folder by checking for the presence of `pyvenv.cfg` there and using it for a similar activation method.

kr-g, 2022-09-19:
what does "hidden" mean in this context ?

can't this not resolved by right-click in file-browser (or by using the little hamburger menu in the right upper corner of that view) view and enable "show hidden files" ?

kr-g, 2022-10-03:
just thinking about ... let's say ... a feature for ... a custom renderer for icons in filebrowser (file: `base_file_browser.py`).

this would require additional checks on:
- for venv - existence of `pyvenv.cfg` inside the folder.
- for python package - existence of `__init__.py` inside the folder.
- for runnable python module - existence of `__main__.py` inside the folder.
- ...

for local file browser i think that would be no problem, but for every remote access/device data retrieval becomes more complex because it requires more separate requests for fetching all that data (for folder entries).
for small devices this is most-likely also a performance issue.

one option could be to have that only with local file browser. 

@aivarannamaa, @Ivoz RFC

aivarannamaa, 2022-10-06:
The idea of activating something hidden doesn't look right to me.

What do you think of an alternative approach -- whenever a (project) directory containing a venv (hidden or not),  is focused in the file browser, the back-end switcher in the lower-right corner automatically gains a new entry corresponding to the venv's executable, so it's easy to explicitly select this interpreter.

kr-g, 2022-10-06:
hidden it is only because it is a dot-file. many developers use such a folder. 
but in principle the logic needs to be bound by testing against the file `pyvenv.cfg` as only constrain.

the idea to have it also in the switcher menu below is good. 
developers used to the switcher would directly find it.
probably directly before the "configure interpreter" (or instead?)

a menu entry to create a fresh, or refresh an existing venv in the current folder with the standard name "venv" would be also nice...

kr-g, 2022-10-06:
i just added a context handler for upgrading a venv.

for running `python3 -m venv --clear venv-path` 
i require the python executable path (of the python version) what comes with thonny itself (because the python in the venv gets wiped...)

how do i get this easily ???

kr-g, 2022-10-07:
@aivarannamaa 

regarding backend-switcher:
here the default local backends (like /usr/bin/python3 ... ) are missing by default.
they appear first when one has such a backend activated before.
the behavior for microcontroller backends seams to be different here