thonny/thonny #2467: Micropython REPL's own auto-completion code function is not working

IlysvlVEizbr, 2022-09-19:
When using Shell in version after 4.0, there is no more Micropython REPL's own auto-completion after pressing Tab button. Now, nothing happens unless there is some hint in the new Thonny's auto-completion function.

The auto-completion is still worked after downgrading the Thonny to 3.3.13.

Expected performance:
```py
>>> asdfasdf = 3
>>> asdf
```
Press Tab, then ↓
```py
>>> asdfasdf = 3
>>> asdfasdf
```
(auto-comple by Micropython itself)

aivarannamaa, 2022-09-19:
Thanks for the report!

Actually, Thonny never used MicroPython's own completion, even in 3.3. This completion only works in the regular REPL mode, but Thonny uses raw-mode.

There is another completion approach introduced in 3.3 and changed in 4.0, but it looks like the newer approach is currently worse than the one in 3.3. I'll try to make it better before the next release.

Hamsanger, 2022-09-21:
Same for me with Thonny version 4.0.1 - I am using the esp32 backend (firmware v1.19.1) and I get no auto-complete suggestions in the REPL after loading a module and then referencing it. For example:
``` py
>>> import dht
>>> a = dht.  <tab> <-- nothing happens
```
I also don't get any auto-complete any more when coding in the editor.

scruss, 2022-10-04:
No tab completion on 4.01 and Pico (MicroPython v1.19.1) in REPL. Editor suggestions completions for keywords and imports, but not methods - and that's where I can really use the tab help for completion

Hamsanger, 2022-10-04:
I have reverted to Thonny version 3.3.13 for comparison and the code completion functions much, much better, both in the REPL and in the code editor including method names (but alas no method argument help).

aivarannamaa, 2023-04-02:
Thonny 4.1 will support name completions in the MicroPython / CircuitPython shell (again).

closed 2023-04-02 by aivarannamaa