thonny/thonny #2470: There is a little “feature” in the interactions of Linux, Thonny and the CircuitPython volume via SSH.

brianlinuxing, 2022-09-21:
There is a little “feature” in the interactions of Linux, Thonny and the CircuitPython volume via SSH.

The solution for Linux users is below.

a) This can be tested in several ways, this is just one:

1. Install the latest version of Thonny on Debian testing (I am using  Xebian).
2. Install CircuitPython 7.3.3 on a Pico, etc.
3. SSH into the machine as a normal user.
4. Bring up Thonny and ensure that it can find the Pico etc.
6. Load a program up from the CIRCUITPY volume and execute.
7. It will, typically, flag up an error (see the screenshots)
[Thonny tries to save the file back before running and hits a minor error.]

b) Issue: The SSH user doesn’t have the capacity to mount/manipulate the volume which causes Thonny problems.

c) **The solution is fairly simple:** in another (non-SSH Window) mount the CIRCUITPY volume in a file manager, after the first error.

d) Verify: re-do test, it passes.

e) The manual way (locally, from the SSH user) doesn't work either:
sudo -i
mkdir /media/CIRCUITPY
mount /dev/sdb1 /media/CIRCUITPY
then invoke Thonny, fails too.

f) Putting the CIRCUITPY volume in fstab, via its UUID, etc doesn’t appear to work, but that might be a more meaningful way.

[Note: Was looking at adding the user to the plugdev group in Debian (logging out, and in again, then testing), but that doesn't fully work, odd as Linux sees it as vfat, nosuid,nodev, etc but the overall ownership is still root.]

[image]
[image]

aivarannamaa, 2022-09-21:
Thank you for this description!

Out of curiosity -- why don't you simply attach your CircuitPython device to the machine running Thonny?

brianlinuxing, 2022-09-21:
It is. The device is plugged straight into the IBM M52, which sits on my desk :)

I then SSH into it,  and bring up Thonny on my other bigger monitor.

I have two machines (a laptop and the old M52 (runs Xebian) next to each other, but prefer to SSH into machines, etc from my main laptop (with the big monitor). I like the client-server approach, habit.  

---
While testing, I reformatted the volume to ext4, which didn't work either (the device is ok but Thonny won't have it)! And gparted won't format FAT16 to anything less than 16Mb, so the solution was to wipe, re-install CircuitPython 7.3.3 firmware again.

DanielJoyce, 2022-12-21:
Add dialout group to ssh-user on machine

`usermod -a -G dialout YOUR_SSSH_USER`

Be sure to use a CAPITAL G or it will set dialout as the primary group for the user. 

May or may not be a security issue, YMMV :)

brianlinuxing, 2022-12-21:
Thank you for the advice, this Linux user was already *in* the dialout group.

Otherwise, it would have not been possible to connect to the device in the first place. :)

(I suspect such errors above relate to the handling of files systems by Thonny and their interactions with Linux, or at least that's what the various issues which have been flagged up in a similar vein suggest, as it is connected to the *file* system.)

aivarannamaa, 2023-01-14:
I'll leave this as is until more users start to use and have trouble with similar setup.

closed 2023-01-14 by aivarannamaa