thonny/thonny #2471: Stop/Restart backend causes file hyperlinks in the traceback to become un-clickable

CosCal, 2022-09-21:
I've noticed a problem in the Stop/Restart backend feature, on both Linux & Windows builds of Thonny v4.0.1.

The blue hyperlink shortcuts, available in the traceback when errors are encountered, were previously working fine in Thonny v3.3.13, but have now been disabled under stop/restart conditions. 

After upgrading to v4.0.1, the Stop/Restart back-end button now triggers all previously working blue file hyperlinks in the traceback to change to all-red text, and they become un-clickable.

This bug or change makes navigation to correct errors much more difficult.
Can the original functionality be restored?

aivarannamaa, 2022-09-25:
Thanks for contacting!

I did change the behavior to reduce the likelihood of confusion, when a stacktrace gets printed, program gets edited and the click on the link now takes the user to the wrong line.

Now I realize that this can happen even when the user process is not closed.

I think I'll revert this decision, like you suggested.

aivarannamaa, 2023-01-14:
In the upcoming version 4.0.2 Thonny won't remove the links anymore.

closed 2023-01-14 by aivarannamaa