thonny/thonny #2472: thonny shell randomly crashed

puppet13th, 2022-09-22:
sometimes thonny shell crashed when i coding in thonny editor.just updated to thonny 4.0.1.the issue exist since thonny 4.0.0.
OS : Windows 10 21H2 x64
```
>>> PROBLEM IN THONNY'S BACK-END: Exception while handling 'get_editor_calltip' (AttributeError: 'CompiledModule' object has no attribute 'non_stub_value_set').
You may need to press "Stop/Restart" or hard-reset your MicroPython device and try again.

See Thonny's backend.log for more info.

Process ended with exit code 1.
```
```
>>> 
Process ended with exit code 3221225725.
```

aivarannamaa, 2022-09-25:
Thank you for the report!

Have you found a way for consistently reproducing this error?

puppet13th, 2022-09-26:
it just happen randomly, the shortest interval is about 10 minutes.i have not tried to restart thonny when it happen.

aivarannamaa, 2023-01-26:
If it happens again, then please send me the backend.log (Tools => Open Thonny data folder), before starting the backend again.

puppet13th, 2023-01-27:
look like there is fault in frontend, please check.
```
>>> 
Process ended with exit code 3221225725.
```

[frontend_faults.log](https://github.com/thonny/thonny/files/10516868/frontend_faults.log)
[backend.log](https://github.com/thonny/thonny/files/10516870/backend.log)

aivarannamaa, 2023-01-27:
Thanks! This is good information. It looks like the `jedi` library has problem in it. I'll investigate this.

Flipje1955, 2023-02-09:
have the same issue recurrently on one of my W10 PC's. Log files available from the crash of this evening, can share if you wish. 
Typically the problem disappears for some time (hrs) after a restart of (at least) Thonny. 
- The suggestion above (deselect restart intepreter before running a script) does not appear to influence the behavior. Thonny 4.0.1 is used in both environments.
- For some time I thought it was related to the version of MP is am using but tonight I used the official 1.19.1 version (and have not seen any mentioning of this issue by others);
- use various different rp2040 boards (but cannot relate the issue to a specific one so appears to be related to the backend.
Problem is easy to solve (start/stop backend), still doing only sw development, so no big deal. 

Should you have a solution ...
thx !

aivarannamaa, 2023-02-10:
@Flipje1955, the logs would be helpful -- please post the backend.log!

Flipje1955, 2023-02-11:
appreciate your efforts ! Hereby, per your request. Let me know if and what you may need further.
[backend.log](https://github.com/thonny/thonny/files/10713463/backend.log)
[frontend_faults.log](https://github.com/thonny/thonny/files/10713464/frontend_faults.log)

Flipje1955, 2023-02-11:
just posted them.
Best regards,

aivarannamaa, 2023-02-12:
@Flipje1955, can you please check, which `jedi` version are you using (Tools => Manage plug-ins)?

Flipje1955, 2023-02-12:
[image: image.png]

per your request...

Flipje1955, 2023-02-12:
[image]

included here again as attached file as the image.png may not be accessible / readable.

aivarannamaa, 2023-04-02:
It looks like the risk of stack overflow is not easy to fix in jedi: https://github.com/davidhalter/jedi/issues/1902

Flipje1955, 2023-04-02:
appreciate your continued attention to the matter; it continues to pop-up (but then, there are worst things in life than this issue). Thank you !

puppet13th, 2023-06-21:
@aivarannamaa i was able to reproduce "exit code 3221225725" with these code :
```
number_list = []
some_list.append( 1 )
some_list.append( 2 )
some_list.append( 3 )
```
the error will occurred when typing "some_list." in the editor

```
MicroPython v1.20.0 on 2023-05-01; LOLIN S3 with ESP32S3

Type "help()" for more information.

>>> 
Process ended with exit code 3221225725.
────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────

MPY: soft reboot
MicroPython v1.20.0 on 2023-05-01; LOLIN S3 with ESP32S3

Type "help()" for more information.

>>> 
Process ended with exit code 3221225725.
────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────

MPY: soft reboot
MicroPython v1.20.0 on 2023-05-01; LOLIN S3 with ESP32S3

Type "help()" for more information.

>>> 
Process ended with exit code 3221225725.
```

[frontend_faults.log](https://github.com/thonny/thonny/files/11812249/frontend_faults.log)
[backend.log](https://github.com/thonny/thonny/files/11812250/backend.log)

currently using thonny 4.1.1

Flipje1955, 2023-06-21:
interesting; the error does not occur when I try your code (using:
MicroPython v1.20.0-162-gd080d427e on 2023-06-06; Raspberry Pi Pico with
RP2040). It gives an error as some_list is not defined.

However, using the same listed MP version, there are moments like yesterday
when the 3221225725 error occurs within say 2 minutes of each other. A
reset of the board, a complete restart of the Thonny environment improves
the situation up to the point that there are days that it does not not
present itself at all. Cannot correlate the exit code to anything specific
I am doing in Thonny.

Using Thonny 4.0.2 on different W10 PC's. Good to see there is a Thonny
4.1.1; will install that soonest.

Thanks + wbr,

aivarannamaa, 2023-06-24:
Thank you for the import! I'll attempt to reproduce this later.

Flipje1955, 2023-06-27:
> Thank you for the input! I'll attempt to reproduce this later.

Just fyi: I upgraded Thonny to 4.1.1 and did some editing on my project
this evening and after say an hour, Thonny aborted with the famous "Process
ended with exit code 3221225725".

Wbr, Rob

HabibMCS, 2023-07-17:
import utime
import s3lcd
import tft_config
import font as font
from machine import reset_cause
tft = tft_config.config( 1,0)
def showscrolltext(char1 , fg = s3lcd.WHITE, bg = s3lcd.RED, pady=60, xs = -1, ys = 0, duration = 40):  
    try:
        tft_config.config( 1,0,backlight = 1)
        tft.init()
        curr = utime.time()
        tft.fill(bg)
        height = tft.height()
        width = tft.width()
        line = height - font.HEIGHT
        while utime.time() - curr < duration:
            for c in char1:
              tft.write(font,c,width - font.WIDTH*2,line-pady,fg,bg,)
              for _ in range(font.HEIGHT // 2):
                    tft.scroll(xs, ys)
                    tft.show()
            utime.sleep(3)
        if reset_cause():
            print(reset_cause())

    finally:
        tft_config.deinit(tft, display_off = True)
        print("scroll completed")
showscrolltext(' Hey its test for display 7809', duration = 5 )
utime.sleep(5)
showscrolltext(' Hey its test for display 7809', duration = 5 )

when the function is called second time the esp32s3 t display crashes.  I don't know whether to put the issue here. I would appreciate if someone helps on this. Thanks in advance

HabibMCS, 2023-07-23:
> ```
> 
> import utime
> import s3lcd
> import tft_config
> import font as font
> from machine import reset_cause
> tft = tft_config.config( 1,0)
> def showscrolltext(char1 , fg = s3lcd.WHITE, bg = s3lcd.RED, pady=60, xs = -1, ys = 0, duration = 40):  
>     try:
>         tft_config.config( 1,0,backlight = 1)
>         tft.init()
>         curr = utime.time()
>         tft.fill(bg)
>         height = tft.height()
>         width = tft.width()
>         line = height - font.HEIGHT
>         while utime.time() - curr < duration:
>             for c in char1:
>               tft.write(font,c,width - font.WIDTH*2,line-pady,fg,bg,)
>               for _ in range(font.HEIGHT // 2):
>                     tft.scroll(xs, ys)
>                     tft.show()
>             utime.sleep(3)
>     finally:
>         tft_config.deinit(tft, display_off = True)
>         print("scroll completed")
> showscrolltext(' Hey its test for display 7809', duration = 5 )
> utime.sleep(5)
> showscrolltext(' Hey its test for display 7809', duration = 5 )
> ```
> 
> when the function is called second time the esp32s3 t display crashes. I don't know whether to put the issue here. I would appreciate if someone helps on this. Thanks in advance

It was solved, when I called the tft instance "tft = tft_config.config( 1,0) " within the function too

aivarannamaa, 2023-08-12:
I now removed some hacks I used together with the Jedi library. Please try https://github.com/thonny/thonny/releases/tag/v4.1.2 and reopen this issue if these errors still occur.

closed 2023-08-12 by aivarannamaa

Flipje1955, 2023-08-13:
thank you for your efforts !

have installed it and will report any results.

Rob

puppet13th, 2023-08-19:
@aivarannamaa i have updated to thonny version 4.1.2 and try to reproduce the issue. i can confirm that the issue fixed in thonny 4.1.2. thank you very much for your hard work.

Flipje1955, 2023-08-19:
Running the upgrade since its release and it is perfectly stable. Thanks
also from my side.
Wbr

adrianblakey, 2024-08-21:
Please could you allow Thonny to copy all file types from a local directory to the board? I want to transfer the files in lib - the copy fails trying to copy mpy files.