thonny/thonny #2473: Can't able to install "Librosa" library on thonny python ide

boAt-Jk, 2022-09-22:


boAt-Jk, 2022-09-22:
/usr/bin/python3 -m pip install --no-cache-dir --user librosa --disable-pip-version-check

Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting librosa
  Downloading librosa-0.9.2-py3-none-any.whl (214 kB)
Collecting decorator>=4.0.10
  Downloading https://www.piwheels.org/simple/decorator/decorator-5.1.1-py3-none-any.whl (10 kB)
Collecting pooch>=1.0
  Downloading https://www.piwheels.org/simple/pooch/pooch-1.6.0-py3-none-any.whl (56 kB)
Requirement already satisfied: joblib>=0.14 in /home/raspberry/.local/lib/python3.9/site-packages (from librosa) (1.2.0)
Collecting resampy>=0.2.2
  Downloading https://www.piwheels.org/simple/resampy/resampy-0.4.2-py3-none-any.whl (3.1 MB)
Requirement already satisfied: scikit-learn>=0.19.1 in /home/raspberry/.local/lib/python3.9/site-packages (from librosa) (1.1.2)
Collecting audioread>=2.1.9
  Downloading https://www.piwheels.org/simple/audioread/audioread-3.0.0-py3-none-any.whl (23 kB)
Requirement already satisfied: packaging>=20.0 in /home/raspberry/.local/lib/python3.9/site-packages (from librosa) (21.3)
Requirement already satisfied: scipy>=1.2.0 in /home/raspberry/.local/lib/python3.9/site-packages (from librosa) (1.8.1)
Requirement already satisfied: numpy>=1.17.0 in /usr/lib/python3/dist-packages (from librosa) (1.19.5)
Collecting numba>=0.45.1
  Downloading https://www.piwheels.org/simple/numba/numba-0.56.2-cp39-cp39-linux_armv7l.whl (3.1 MB)
Collecting soundfile>=0.10.2
  Downloading https://www.piwheels.org/simple/soundfile/SoundFile-0.10.3.post1-py2.py3-none-any.whl (21 kB)
Collecting llvmlite<0.40,>=0.39.0dev0
  Downloading llvmlite-0.39.1.tar.gz (132 kB)
Requirement already satisfied: setuptools<60 in /usr/lib/python3/dist-packages (from numba>=0.45.1->librosa) (52.0.0)
Requirement already satisfied: pyparsing!=3.0.5,>=2.0.2 in /home/raspberry/.local/lib/python3.9/site-packages (from packaging>=20.0->librosa) (3.0.9)
Collecting appdirs>=1.3.0
  Downloading https://www.piwheels.org/simple/appdirs/appdirs-1.4.4-py2.py3-none-any.whl (9.6 kB)
Requirement already satisfied: requests>=2.19.0 in /usr/lib/python3/dist-packages (from pooch>=1.0->librosa) (2.25.1)
Requirement already satisfied: threadpoolctl>=2.0.0 in /home/raspberry/.local/lib/python3.9/site-packages (from scikit-learn>=0.19.1->librosa) (3.1.0)
Requirement already satisfied: cffi>=1.0 in /home/raspberry/.local/lib/python3.9/site-packages (from soundfile>=0.10.2->librosa) (1.15.1)
Requirement already satisfied: pycparser in /home/raspberry/.local/lib/python3.9/site-packages (from cffi>=1.0->soundfile>=0.10.2->librosa) (2.21)
Building wheels for collected packages: llvmlite
  Building wheel for llvmlite (setup.py): started
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-k9e8jono/llvmlite_be6b71c1fbf0446f9790f8f13c639f21/setup.py'"'"'; __file__='"'"'/tmp/pip-install-k9e8jono/llvmlite_be6b71c1fbf0446f9790f8f13c639f21/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-v_qx4abc
       cwd: /tmp/pip-install-k9e8jono/llvmlite_be6b71c1fbf0446f9790f8f13c639f21/
  Complete output (11 lines):
  running bdist_wheel
  /usr/bin/python3 /tmp/pip-install-k9e8jono/llvmlite_be6b71c1fbf0446f9790f8f13c639f21/ffi/build.py
  LLVM version... Traceback (most recent call last):
    File "/tmp/pip-install-k9e8jono/llvmlite_be6b71c1fbf0446f9790f8f13c639f21/ffi/build.py", line 226, in <module>
      main()
    File "/tmp/pip-install-k9e8jono/llvmlite_be6b71c1fbf0446f9790f8f13c639f21/ffi/build.py", line 216, in main
      main_posix('linux', '.so')
    File "/tmp/pip-install-k9e8jono/llvmlite_be6b71c1fbf0446f9790f8f13c639f21/ffi/build.py", line 135, in main_posix
      raise RuntimeError(msg) from None
  RuntimeError: Could not find a `llvm-config` binary. There are a number of reasons this could occur, please see: https://llvmlite.readthedocs.io/en/latest/admin-guide/install.html#using-pip for help.
  error: command '/usr/bin/python3' failed with exit code 1
  ----------------------------------------
  Building wheel for llvmlite (setup.py): finished with status 'error'
  ERROR: Failed building wheel for llvmlite
  Running setup.py clean for llvmlite
Failed to build llvmlite
Installing collected packages: llvmlite, numba, appdirs, soundfile, resampy, pooch, decorator, audioread, librosa
    Running setup.py install for llvmlite: started
    Running setup.py install for llvmlite: finished with status 'error'
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-k9e8jono/llvmlite_be6b71c1fbf0446f9790f8f13c639f21/setup.py'"'"'; __file__='"'"'/tmp/pip-install-k9e8jono/llvmlite_be6b71c1fbf0446f9790f8f13c639f21/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-zyyz9bvd/install-record.txt --single-version-externally-managed --user --prefix= --compile --install-headers /home/raspberry/.local/include/python3.9/llvmlite
         cwd: /tmp/pip-install-k9e8jono/llvmlite_be6b71c1fbf0446f9790f8f13c639f21/
    Complete output (14 lines):
    running install
    running build
[... cut ...] p/pip-install-k9e8jono/llvmlite_be6b71c1fbf0446f9790f8f13c639f21/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-zyyz9bvd/install-record.txt --single-version-externally-managed --user --prefix= --compile --install-headers /home/raspberry/.local/include/python3.9/llvmlite Check the logs for full command output.
Process returned with code 1

aivarannamaa, 2022-09-25:
It doesn't look like Thonny specific error. I recommend contacting with the developer of the library.

closed 2022-09-25 by aivarannamaa