thonny/thonny #2474: Use external editor in Thonny - Any plans ?

ron-deboer, 2022-09-26:
Any plans to implement "Use External Editor" in Thonny ?

Thonny is a really great app, however the internal editor is a bit clucky and primitive.

brianlinuxing, 2022-09-26:
I think it is a good idea, but could it be implemented by simply calling a local shell and invoking an editor etc?

aivarannamaa, 2022-09-29:
Do you mean editing in files on local disk? In this case, what would you use Thonny for? If you describe your desired flow then I may be able to suggest something.

datamafia, 2023-01-07:
This is an essential feature that I would place a high value on.

My workflow in Arduino is the perfect example.  I edit in my favorite C IDE (JetBrains Lion).  I use the default/free Arduino editor to connect/debug/ship.  This operation is seamless via a bool setting in the Arduino IDE preferences.  Under the hood, the app is forced to trust files despite changes and handle caching carefully.  Additionally, any autosave must be disabled.

# Why is this important?

Thonny is a great tool, and it has its place.  [for me] That place is an easy and reliable way to connect to something like a Rasp Pi Pico.  

In the case of using Thonny for my Pico, I am in my favorite IDE, and I am getting unnecessary dialogs regarding changes on file (part of the IDEs file watching and alert system).  

Having used the same IDE for over a decade, collecting macros and shortcuts, and writing a million lines of code (so far), my productivity is squarely rooted in that IDE; even the config has been under version control the entire time. 

Regardless, thank you for this IDE. I sincerely appreciate the work.

datamafia, 2023-01-10:
@aivarannamaa  Thanks for the response, I will examine mpremote (I was considering writing a serial connection/logger to solve this problem)

Regarding this feature - the "competing IDE" enables Python (or whatever) pro's to work as expected.  I am a Career Python developer (15 years in 2023), and I have a well-established way of working. Thonny is excellent, and 101% respect for the creators and effort, but all I need is terminal output and upload/run on the device.

RPI and Arduino are mostly tinkerers.  They will not spend hundreds of dollars a year on professional tools.  Since this is my job, I have the tools and see no reason to impede my development.

Again thanks for the tools.

closed 2023-01-28 by aivarannamaa