thonny/thonny #2476: Save As doesn't include .py extension

bfletcherjohnson, 2022-09-27:
When saving code for the first time, even though Save as type: shows 'Python files(*.py; *.pyw; *.pyi) the file is saved without the type extension.  It would be nice for the file to be saved with the .py extension appended.  The extension must be added manually in the File name: window.

aivarannamaa, 2022-09-29:
Which OS and which Thonny version are you using?

bfletcherjohnson, 2022-09-29:
Sorry:

Thonny 4.0.1

Windows 10 (64-bit)

Python 3.10.6 (64-bit)

Tk 8.6.12

aivarannamaa, 2022-09-30:
That's weird, I can't reproduce this with the same setup -- when I write "foo" as file name, I'll end up with "foo.py". Can you post the screenshot of your Save As dialog just before clicking "Save"? Maybe I can spot some difference compared to my system.

bfletcherjohnson, 2022-09-30:
Thanks for your attention.  I think I understand the behavior.

 

I was saving files with a period in the filename.

 

If there is a period in the filename, the .py is NOT appended

 

Anything without a period in the filename works as expected, the “.py” is appended to the filename.

 

“Section_18.1_test_save”  will be saved as “Section_18.1_test_save”; the “.py” will NOT be appended.

“Section_18_1_test_save” will be saved as  “Section_18_1_test_save.py”; the “.py” is appended as expected

 

Not sure if that constitutes a bug. Perhaps I should not be saving python programs with a period in the filename.

aivarannamaa, 2022-10-01:
At the moment Thonny thinks the user already provided an extension. I'd rather not make this more complicated. You can keep using periods in the file name but in this case you need to provide the `.py` extension explicitly.

bfletcherjohnson, 2022-10-01:
Understood, thanks again.

closed 2023-01-28 by aivarannamaa