thonny/thonny #2477: ctrl-shift-arrow selection jumps too far

pmbdk, 2022-09-27:
When I edit in Thonny (4.0.1, windows 10 64-bit, python 3.10.6 64-bit, Tk 8.6.12), the ctrl-shift-arrow selection methods works differently than most other editors. 

Assuming I have the following code:
```python
class SomeClass:
      def some_method():
            pass
```
and I place my cursor right before "class". 

The first time I press ctrl-shift-right arrow, I select "class" which is what I would expect. However, the next time I do it, I select down to right before "def", which is contrary to what other editors do (such as Notepad++, vscode, pycharm ,etc.) which is to only select up to right before the ":" after "SomeClass".

The same goes for ctrl-arrow. 

Is there anything which can be done to change this behaviour?

aivarannamaa, 2022-09-29:
Thank you for the report!

I'll try to reproduce this on a Windows computer later, but on Linux, it works according to your expectations.

aivarannamaa, 2023-01-08:
I could reproduce this in Windows. This behavior (including the difference in Windows and Linux) comes from Thonny's UI toolkit (Tk). I'll see later whether I can do something about this. If not, then we'll need to wait for the next Tk version.