thonny/thonny #2478: Thonny, Pipkin and local MicroPython package management, is it possible?

brianlinuxing, 2022-09-27:
Is there any way to allow a local version of MicroPython to have the package management facilities, as exists with the Pico etc

Scenario A:

Running Thonny on a networked Pico, the Manage packages option is available and works well. I can install the certifi package without any issues.

Scenario B:

Running Thonny on a locally build version of MicroPython,  the Manage packages option is not available.

If I use upip, instead, to install ceftifi directly it fails.

Reasoning:

I want to test code locally under the Linux version of MicroPython, calculate package sizes, fix code then migrate to the Pico device. 

But for that to work well I need package management on the local Linux version of MicroPython to work consistently. 

Upip will work with some modules and fail with others, however, when *in* the Thonny package management is much better.

[The images upip_v4 abd upip_v5 show Thonny working on a local Linux version. upip_v6 is Thonny connected to the networked Pico.]

Question:

**Is there a trick to invoking pipkin either inside MicroPython on Linux or getting it to work well with the /.micropython/lib structure to put modules there?**

PS: I know some of the packages aren’t meant to work with MicroPython and I will fix that code as it goes along, but firstly I would like to get the modules into the  /.micropython/lib structure without too much hassle. 

I would welcome helpful suggestions. 

Thanks, Brian.

[image]
[image]
[image]

brianlinuxing, 2022-09-28:
I had an idea, but I'd welcome suggestions from @aivarannamaa

1. I’d called pipkin locally (everything is running in a VM) and get it to pull over what I needed, outside of Thonny.

It nearly seems to work, except for an odd error about pip (which is installed elsewhere, but I am probably missing something)

The command executed: **/home/brian/.local/bin/pipkin -v --dir /home/brian/.micropython/lib install micropython-logging**

2. The output suggests either an error message should come up and say "this bit is missing" or have a suggested method of rectifying it: 

brian@SparkyuPython3:~$ /home/brian/.local/bin/pipkin -v --dir /home/brian/.micropython/lib install micropython-logging

**The output:**

Starting install
Start populating venv
Using existing working environment at /home/brian/.cache/pipkin/workspaces/77b23d5cc84289b84f594d790c14a557
Received lock on the working environment
Querying site packages path for /home/brian/.cache/pipkin/workspaces/77b23d5cc84289b84f594d790c14a557
Got site packages path /home/brian/.cache/pipkin/workspaces/77b23d5cc84289b84f594d790c14a557/lib/python3.10/site-packages
Clearing /home/brian/.cache/pipkin/workspaces/77b23d5cc84289b84f594d790c14a557/lib/python3.10/site-packages
Done populating venv
Querying site packages path for /home/brian/.cache/pipkin/workspaces/77b23d5cc84289b84f594d790c14a557
Got site packages path /home/brian/.cache/pipkin/workspaces/77b23d5cc84289b84f594d790c14a557/lib/python3.10/site-packages
Using PipkinProxy at http://127.0.0.1:36628
Calling pip: /home/brian/.cache/pipkin/workspaces/77b23d5cc84289b84f594d790c14a557/bin/python3 -I -m pip --disable-pip-version-check --trusted-host 127.0.0.1 install --no-compile --use-pep517 --upgrade-strategy only-if-needed micropython-logging --index-url http://127.0.0.1:36628
/home/brian/.cache/pipkin/workspaces/77b23d5cc84289b84f594d790c14a557/bin/python3: No module named pip

3. I checked $PATH just in case something was missing but seems OK:
echo "${PATH//:/$'\n'}"
/usr/local/bin
/usr/bin
/bin
/usr/local/games
/usr/games
/home/brian/.local/bin

aivarannamaa, 2022-09-29:
Thank you for this topic!

Actually, Pipkin already has placeholders for supporting local (and remote over SSH) Unix MicroPython target, eg: https://github.com/aivarannamaa/pipkin/blob/afca9bccc41707a13795ca17148d3914ded063b1/pipkin/adapters.py#L293

I just created an explicit issue as a reminder for this: https://github.com/aivarannamaa/pipkin/issues/3 

Once this part gets solved, I'll enable this in Thonny as well.

brianlinuxing, 2022-09-29:
Thank you very much. :)

brianlinuxing, 2022-09-30:
BTW, when you are closer to a solution I am very, very happy to test it on the Linux side. before release :)