thonny/thonny #2482: [macOS] Confusing error message when user doesn't grant permission to Desktop or Downloads

kcarnold, 2022-09-29:
Situation: student writes a script, saves it on Desktop, hits Run, and Thonny gives a confusing error message that suggests checking the backend log. Nothing obvious shows up there, but hidden in the frontend log is a "File not found", which eventually leads to the root cause: they hadn't allowed Thonny permission to access the Desktop. Granting permission in System Preferences -> Privacy -> Files and Folders solves the problem.

I wonder if there's a way to detect this situation and offer a better error message.

aivarannamaa, 2022-09-30:
Thank you for the report!

I thought this permission problem is a history after I introduced pkg-installer. It appeared with Catalina and it was really hard to reproduce. 

Can you give me the content of the backend.log?

aivarannamaa, 2023-01-28:
Please reopen if you experience this again and have more information to add.

closed 2023-01-28 by aivarannamaa

kcarnold, 2026-01-22:
I saw this again in lab today and was able to reproduce it on my own computer by denying the Thonny app access to my Documents (in "System Settings" -> "Privacy & Security" -> "Files & Folders") — and saving a file *in a subfolder of Documents*. Then the Shell says:

```
PROBLEM IN THONNY'S BACK-END: mainloop error (PermissionError: [Errno 1] Operation not permitted).
See Thonny's backend.log for more info.

Process ended with exit code 17
```

And backend.log says:

```
13:20:16.945 [MainThread] INFO    thonny: TIME/MODS 20.507 Done creating executor 
13:20:16.947 [MainThread] INFO    thonny: TIME/MODS 0.002 Sending message ToplevelResponse 
13:20:19.693 [MainThread] INFO    thonny: TIME/MODS 2.746 Done creating executor 
13:20:19.697 [MainThread] INFO    thonny: TIME/MODS 0.004 Sending message ProgramOutput 
13:20:19.698 [MainThread] INFO    thonny: TIME/MODS 0.000 Sending message ProgramOutput 
13:20:19.698 [MainThread] INFO    thonny: TIME/MODS 0.000 Sending message ProgramOutput 
13:20:19.698 [MainThread] INFO    thonny: TIME/MODS 0.000 Sending message ProgramOutput 
13:20:19.698 [MainThread] INFO    thonny: TIME/MODS 0.000 Sending message ToplevelResponse 
13:20:25.341 [MainThread] INFO    thonny: TIME/MODS 5.642 Sending message highlight_occurrences_response (+1 modules)
13:20:25.341 [MainThread] INFO    thonny: NEW MODS ['thonny.jedi_utils']
13:20:34.853 [MainThread] INFO    thonny: TIME/MODS 9.512 Sending message highlight_occurrences_response 
13:20:36.668 [MainThread] INFO    thonny: TIME/MODS 1.815 Sending message ToplevelResponse 
13:20:36.669 [MainThread] ERROR   thonny.backend: mainloop error
Traceback (most recent call last):
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/backend.py", line 87, in mainloop
    self._handle_normal_command(msg)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/cpython_backend/cp_back.py", line 273, in _handle_normal_command
    super()._handle_normal_command(cmd)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/backend.py", line 311, in _handle_normal_command
    self.send_message(real_response)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/cpython_backend/cp_back.py", line 911, in send_message
    msg["cwd"] = os.getcwd()
PermissionError: [Errno 1] Operation not permitted
13:20:36.671 [MainThread] INFO    thonny: TIME/MODS 0.003 Sending message ProgramOutput 
13:20:36.671 [MainThread] INFO    thonny: TIME/MODS 0.000 Sending message ProgramOutput 
13:20:36.671 [MainThread] INFO    thonny.backend: After mainloop
```

This is 4.1.7; it looks like there hasn't been a release in a while, so I'm having trouble matching that up with current GitHub code.

If you don't use a subdir of the denied folder (so `os.getcwd()` succeeds), you instead get a popup (still pretty cryptic) with:

```
Traceback (most recent call last):
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/workbench.py", line 1794, in event_generate
    handler(event)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/assistance.py", line 143, in handle_toplevel_response
    self._start_program_analyses(
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/assistance.py", line 277, in _start_program_analyses
    analyzer.start_analysis(main_file_path, imported_file_paths)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/misc_analyzers.py", line 227, in start_analysis
    self.completion_handler(self, list(self._get_warnings(main_file_path)))
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/misc_analyzers.py", line 239, in _get_warnings
    for item in os.listdir(main_file_dir):
PermissionError: [Errno 1] Operation not permitted: '/Users/ka37/Documents'
```