thonny/thonny #2483: Thonny freeze with some long list

nsi-xyz, 2022-09-30:
I have a 96 KB python file that uses lists of 2000 values.

Each time I add a list of 2000 values ​​in the script editor, thonny is slower and slower, he freezes a bit, then more and more often until it makes typing in the IDE very difficult:

- Cursor no longer displayed
- IDE crashes
- The scrolling is done offset
- (...)

On spyder I don't encounter this problem, but spyder is more complicated to install and in some ways less intuitive, so less easy for students to learn.

[pyka3d_lags.zip](https://github.com/thonny/thonny/files/9685206/pyka3d_lags.zip)

aivarannamaa, 2022-09-30:
Thank you for the report!

Thonny uses Tk's Text widget for code editor and unfortunately it is very slow with long lines. You should get much better performance if you insert some linebreaks into your list literal.

BTW. I opened your file in the Gedit text editor in Ubuntu, and even this program was struggling with your long line.

nsi-xyz, 2022-10-01:
Thanks for such a quick response.

I understood that it is not a bug because not correctable, but related to the text editor used.

I tried before reporting my problem to change these settings, passing that I could work around the problem but without success.

[image]

I would have tried to replace the 500 by 79 (PEP 8 suggests lines should be limited to 79 characters) to see if that improves responsiveness but it is impossible to go below 500.

aivarannamaa, 2022-10-01:
These options affect the presentation of the program output, not the program source code in the editor. You need to break lines manually, with ENTER key.

closed 2023-01-28 by aivarannamaa