thonny/thonny #2484: Is it possible to flash other binaries (Tasmota..) via Thonny?

hagenbuch, 2022-09-30:
I'm just asking because it would be so convenient...

aivarannamaa, 2022-09-30:
I assume you mean whether some of Thonny's MicroPython or CircuitPython flashing dialogs could also be used for installing something else than MicroPython / CircuitPython. 

I suppose the ESP dialogs already can be used for this, provided the correct target address for Tasmota is the same as for MicroPython.

If this is not the case then I don't intend to do anything about this. I want Thonny to remain a Python IDE for beginners.

hagenbuch, 2022-09-30:
Many thanks for your swift answer! 

I also don't want to change Thonny, it would just be great to use that tool for more than intended. Maybe someone knows the answer for the address question. I might as well give it a try :) apparently the bootloder of an ESP32 is in ROM and can't be changed, so I guess we can't brick it.

Jason2866, 2022-11-24:
@hagenbuch You can not brick a esp32 as long you do not burn efuses (which needs extra effort to do). So just try. The default address for the firmware to start is 0x10000
Tasmota firmware is not address hard coded, so it can be flashed anywhere. So if the installed bootloader expects to start on a different address just flash there.
BUT you may run later in problems with Tasmota when the used partition scheme is to different from the standard Tasmota partition scheme.

closed 2022-12-28 by aivarannamaa