thonny/thonny #2485: PROBLEM IN THONNY'S BACK-END: Exception while handling 'Run' (ConnectionError: EOF).

rstmsvk, 2022-10-01:
I am trying to run an example code to test wifi but I am getting an error like this. Here is the screenshot of the whole window. What is the problem? Thank you for your time.
[image]

aivarannamaa, 2022-10-01:
It looks like the device or MicroPython interpreter had a reset. What's in your boot.py ?

rstmsvk, 2022-10-01:
It's empty. I just uploaded the firmware. I have tried some other interpreters and got a similar error. Also, I check the board with Arduino ide in order to see if it was broken or not and the board was fine. It is working fine when I use Arduino ide.

aivarannamaa, 2022-10-02:
Which exact device are you using? Does it make a difference if you uncheck "Restart interpreter before running a script" at interpteter configuration page (click on the intepreter switcher at the lower-right corner of Thonny's main window and select "Configure interpreter").

rstmsvk, 2022-10-02:
I am using ESP32-WROOM-32D DevkitC-V4. I unchecked it, and I am still getting the same error, unfortunately.

springjools, 2022-10-04:
I get this too with a raspberry pi pico WH

rstmsvk, 2022-10-13:
I got an esp8266 now and it is running fine. I think the one that I have is broken.

aivarannamaa, 2023-01-14:
Closing as I'm not able to reproduce. Please reopen if you have more clues to add.

closed 2023-01-14 by aivarannamaa

Mariopolonia0, 2024-09-07:
i have the problem