thonny/thonny #2489: tkinter code works in IDLE but not Thonny.

bfletcherjohnson, 2022-10-04:
Running Thonny 4.0.1 in Windows 10 (64-bit) using Python 3.10.6 (64-bit).

This two-line line code:
import tkinter as tk
window = tk.Tk()

does NOT work (produce a small window on computer screen) in Thonny when
- run in Shell
- run as a .py file

It does work in IDLE when
- run in shell
- run as a .py file

?

bfletcherjohnson, 2022-10-04:
Hmm, I see adding a third line "window.mainloop()" is necessary for Thonny, but not for IDLE?

aivarannamaa, 2022-10-06:
Thank you for the report!

In general, calling `mainloop()` is required for Tkinter programs. Both IDLE and Thonny have implemented some tricks to allow interactive implementation of Tkinter programs, but it looks like it doesn't work very well in Thonny.

I'll investigate this under #2391

closed 2022-10-06 by aivarannamaa