thonny/thonny #2491: Name logging not defined in mypy plugin __init__.py line 110

solsword, 2022-10-05:
See the PR comments for more info; a student ran into this and I've issued PR #2490 with the trivial fix of importing the missing function. I was able to trigger this by saving the following file as test.py and attempting to run it:

```py
print('hi')

import test
```

kr-g, 2022-10-06:
can you please provide some more information here.
like your students source code, and expected behavior or print output.

one remark:
there is standard test package ->
https://docs.python.org/3/library/test.html#module-test

so it is not useful to name an own python file also `test.py`
your code above will import itself again - i assume.

kr-g, 2022-10-06:
sorry my mistake, just saw it is a typo error in the code "logging" instead of "logger".

-> should be ->
`logger.error("Can't parse MyPy line: " + line.strip())`

the extra import of error (or error as log_error), please do remove.

can you change the code fix and test again before submitting, thanks.

kr-g, 2022-10-06:
https://github.com/thonny/thonny/pull/2490

solsword, 2022-10-06:
Thanks for the quick response here. I didn't read the code enough to see that there was a logger variable and realize that `logging` was just a typo, since I was trying to fix things for the student quickly and then get home. I've updated the PR to fix this by changing logging -> logger (it still gets rid of the unused import sys I think).

aivarannamaa, 2022-10-07:
@solsword, thanks for the report and fix and @kr-g, thanks for assistance! I'll try to find a fix for the parsing error under #2495

closed 2022-10-07 by aivarannamaa