thonny/thonny #2492: Add a signature when running applications in Thonny IDE

jmparatte, 2022-10-06:
Hi,
I'm concerned with identify when running applications in Thonny IDE.
For example when running applications interacting with console input(), to switch input() to something like reading a file.

I propose inserting a line in file **C:\Thonny\Lib\site-packages\thonny\plugins\micropython\mp_back.py** like that:
`         self._prepare_rtc()` **line 157**
`         self._execute("THONNY_IDE={'name': 'Thonny', 'version': '4.0.1'}") # Thonny IDE signature` **inserted line**
`         self._send_ready_message()` **line 158**
So, I can check if running with Thonny IDE like that:
`    try:`
`        if THONNY_IDE: pass`
`        # Thonny IDE detected`
`        ...` 
`    except:`
`        # Thonny IDE not detected`
`        ...`

jmP

kr-g, 2022-10-06:
i m not sure if i understood your requirement, or use-case, properly.

but probably checking in your code against the platform, or implementation, solves this better together with that `try` block.

with micropython
```
>>> import sys
>>> sys.platform
'esp8266'
```

with linux
```
>>> import sys
>>> sys.platform
'linux'

```

another option cloud be checking against this

```
import sys
sys.implementation

```
or

```
>>> import os
>>> os.uname()

```

please note that not all of the above is available on micropython.
you need to try what is there and fits best.

jmparatte, 2022-10-06:
Hi, thanks for your reply, but my request is about running in or out of Thonny, to take some special actions if Thonny detected, not to check the implementation or platform, not to check Python or Micropython.

kr-g, 2022-10-06:
run this
```
import os
for k, v in os.environ.items():
    print(k, v)

```

at the end you will find

```
THONNY_USER_DIR /home/benutzer/repo/thonny/.venv/.thonny
THONNY_FRONTEND_SYS_PATH ['/home/benutzer/repo/thonny', '/usr/lib/python38.zip', '/usr/lib/python3.8', '/usr/lib/python3.8/lib-dynload', '/home/benutzer/repo/thonny/.venv/lib/python3.8/site-packages', '/home/benutzer/repo/thonny/.venv/.thonny/plugins/lib/python3.8/site-packages']
THONNY_LANGUAGE en_US

```

thonny sets those variables, otherwise they will be not there.
works? fine?

kr-g, 2022-10-07:
@aivarannamaa 

how about having an addtional env var `THONNY_VERSION` ?

https://github.com/thonny/thonny/blob/b6330097bea96343b01c68e6a069a756fc0e5e42/thonny/running.py#L982

jmparatte, 2022-10-07:
Yes, you are right for any Python platform, I check `THONNY_USER_DIR` with succces with Win32, MacOS, Linux, RPi platforms. But with Micropython ESP32, that's doesn't work. I currently use this trick `thonny_detected = time.time()>31*86400` with ESP32.

kr-g, 2022-10-07:
> thonny_detected = time.time()>31*86400

this looks more like an esoteric thing to me.
RFC

aivarannamaa, 2022-10-07:
@jmparatte, what about my proposal at https://github.com/micropython/micropython/discussions/9245#discussioncomment-3678772 ? Doesn't it cover your needs?

aivarannamaa, 2023-01-28:
Fixed by #2494

closed 2023-01-28 by aivarannamaa