thonny/thonny #2498: Links to support Ukraine?

avovk1, 2022-10-10:
I live in Kyiv, and I came by support link. I see one not good scene - you mention international red cross, which not once became center of scandal and work _tightly_ with occupant to hide real situation, like in Olenivka - still no info about prisoners of war who was killed there.
I think, it would be better idea to place link to Ukrainian Red Cross, not connected to international one. ([https://redcross.org.ua/en/](https://redcross.org.ua/en/) is the home, and donation outside of Ukraine is [https://donate.redcrossredcrescent.org/ua/donate2/~my-donation](https://donate.redcrossredcrescent.org/ua/donate2/~my-donation))
Also, you could place there link to "Come back alive", as one of the biggest if not one. ([https://savelife.in.ua/en/](https://savelife.in.ua/en/) - home page, [https://savelife.in.ua/en/donate-en/#donate-army-card-monthly](https://savelife.in.ua/en/donate-en/#donate-army-card-monthly) - donation)

About second link I have no questions, and thank you for such a good software, that I use regularly)
Hope that last empire in the world will end here, not further.

aivarannamaa, 2022-10-10:
Thanks for this information! I now swapped the Red Cross link and added savelife one.

closed 2022-10-10 by aivarannamaa

xkonglong, 2023-02-03:
I hope you can also support Syria refugees, Afghanistan refugees and all countries invaded by the United States. Thank you.