thonny/thonny #2499: Edited Codefile not automatically saved

tb3456, 2022-10-10:
In Version 3.xx Code was automatically saved to file when Code was executed.

In Version 4.01 this doesn't happen anymore.

There should be an Option to select that behaviour.

aivarannamaa, 2022-10-12:
Thanks for the suggestion!

I don't mind adding an option for this. I imagine a checkbox labelled "Allow running unsaved files" which is checked by default. Please note that automatic saving of named files still happens in Thonny 4.

tb3456, 2022-10-14:
Hi Aivar,

thank you very much for your fast reaction!

I'm very satisfied with Thonny and like to use it for Python trainings 
since several years, because it's OS independent, very easy to use but 
sufficent enough, and to don't prefer any of the different IDEs the 
participants use (Visual Code, PyCharm, Anaconda, Spyder, ...).

But the new version 4.XX seems to have some (new) problems/bugs and/or 
seems to have a different behaviour than version 3.XX, e.g.

* Doesn't force a file name for new scripts during first run.

* Doesn't always save named files if script is run after changes.

* Shortcut key "F5" doesn't work the same way as selection menu "Run - 
Run current script" or hitting "Run"-Symbol.

* Restart needed several times during a day because doesn't react anymore.

I'm using Thonny on a MacBook Pro 16" (2019) / 2.3 GHz i9 8-Core / 64 GB 
RAM / 2 TB disk / macOS Monterey Version 12.6

A new option I would like propose is:

* Switch on/off "-O" (optimize) for switching on/of "assert/__debug__" 
functionality.

Hope this will clarify the issue!

Thanks
Thomas Birnthaler

> Thanks for the suggestion!
>
> I don't mind adding an option for this. I imagine a checkbox labelled 
> "Allow running unsaved files" which is checked by default. Please note 
> that automatic saving of named files still happens in Thonny 4.

aivarannamaa, 2022-10-18:
> Doesn't always save named files if script is run after changes.

Can you tell me how to reproduce this?

> Shortcut key "F5" doesn't work the same way as selection menu "Run -
Run current script" or hitting "Run"-Symbol.

Could you elaborate?

> Switch on/off "-O" (optimize) for switching on/of "assert/__debug__"
functionality.

I'd rather not add this to a beginner IDE.

aivarannamaa, 2023-07-10:
The option for requiring saved files is now added.

closed 2023-07-10 by aivarannamaa