thonny/thonny #2500: Cannot open Thonny

thelmuth, 2022-10-11:
I teach a Python class and have my students use Thonny. One student was using Thonny yesterday, and it was working fine, and then today when she opened Thonny, it wouldn't open and crashes with a "Python quit unexpectedly" error. 

**Details:**
- macOS Catalina
- Version 10.15.7
- 2020 MacBook Pro

The student says that she didn't change anything about her installation, her operating system, or her system settings between when it worked and now. I suspect something changed, but it's possible nothing did. She says the most recent things she did were to install NLTK using Thonny's package manager.

**Error:** 
[thonny_crash.txt](https://github.com/thonny/thonny/files/9759091/python_crash.txt)

**Attempts to fix:**
I tried reinstalling Thonny using the pkg, then tried using pip3, and then tried using homebrew. The first and third gave the same error, and the pip3 installation failed with a different error.

aivarannamaa, 2022-10-12:
Thanks for the report!

This crash seems to come from Tkinter and I'm afraid I can't fix this. It's worth trying to find the reason, though. The first thing I'd try, is temporarily removing Thonny's user directory `~/Library/Thonny`. If this reliably cures the problem, then I'd be glad to investigate the contents of this folder.

thelmuth, 2022-10-18:
This worked! Thank you!

I've attached the `~/Library/Thonny` directory, if you want to investigate what was going on.

[ThonnyBACKUP.zip](https://github.com/thonny/thonny/files/9813951/ThonnyBACKUP.zip)

aivarannamaa, 2022-10-18:
Thanks!

The frontend_faults.log was a lot more readable than macOS's system crash log. It turned out the crash occurred at the moment when Thonny showed a messagebox complaining about bad encoding in the source file to be reopened. I'll try to reproduce this on macOS later.