thonny/thonny #2502: Unable to access wxpython

wxmurph, 2022-10-13:
After installing Windows 11 my wxpython code stopped working.  I tried to download with Pypi from the tools menu in Thonny and it give the ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output. Error: process returned with code 1.

I then downloaded the following from command line:
Python 3.10.8
camelcase version 0.2
mysql-connector-python version 8.0.28
numpy version 1.23.4
Pillow version 9.2.0
pip version 22.2.2
pygame version 2.1.2
setuptools version 63.2.0
six version 1.16.0
wheel version 0.37.1
wxPython version 4.2.0

They all work in the command prompt window but will not work in Thonny.  

I get the following error when I run the simple Hello World program: 

# Hello World wxPython

import wx # import wxPython package

app = wx.App() # creates an application object

frm = wx.Frame(None, title="Hello World")

frm.Show()

app.MainLoop() # Starts the app 

Error:
Python 3.7.9 (bundled)
>>> %Run '1A. Hello World wxPython.py'
Traceback (most recent call last):
  File "C:\Users\Admin\ThonnyProjects\wxPython NewBoston\1A. Hello World wxPython.py", line 3, in <module>
    import wx # import wxPython package
ModuleNotFoundError: No module named 'wx'
>>> 
I noticed above that the program is not running the current version of Python, but I don't know how to fix it.  Please help!

wxmurph, 2022-10-13:
I downloaded the latest version of Thonny, it allowed me to download the wxpython files on pypi Thonny still does not recognize Import wx nor will it run wxpython.

aivarannamaa, 2022-10-18:
Thank you for the report!

Did you install wxpython via "Tools => Manage packages" or "Tools => Manage plug-ins" ? The former is the right one.

aivarannamaa, 2023-01-28:
Please reopen if you still have this problem and can add more information.

closed 2023-01-28 by aivarannamaa